google/aiyprojects-raspbian #336: aiyprojects-2018-04-13 'Check Cloud' provided URL gets a 404 'Not Found'

juggledad, 2018-04-22:
I installed the aiyprojects-2018-04-13.img on a RPI3. When I run the 'Check Cloud' script it says:

> Please follow these instructions to get Google Cloud credentials:
> https://cloud.google.com/speech/docs/getting-started#set_up_your_project
> and save them to /home/pi/cloud_speech.json

however that URL produces a 404 error

juggledad, 2018-04-23:
Well you still need to fix the difference is which name to call the json file.

In addition, I click the 'Start dev terminal' and the V1 guide says to enter:

> src/examples/voice-v1/assistant_library_demo.py

However there is no voice-v1 folder and if I use 

> src/examples/voice/assistant_library_demo.py
it fails with:
```
pi@voice:~/AIY-projects-python $ src/examples/voice/assistant_library_demo.py
Traceback (most recent call last):
  File "src/examples/voice/assistant_library_demo.py", line 73, in <module>
    main()
  File "src/examples/voice/assistant_library_demo.py", line 67, in main
    with Assistant(credentials) as assistant:
  File "/opt/aiy/projects-python/src/aiy/assistant/library.py", line 33, in __init__
    self._model_id = device_helpers.register_model_id(credentials)
  File "/opt/aiy/projects-python/src/aiy/assistant/device_helpers.py", line 93, in register_model_id
    r.raise_for_status()
  File "/usr/local/lib/python3.5/dist-packages/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Forbidden for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/voice-1-201708/deviceModels/voice-1-201708-voice-kit

```
 
Did anyone text the latest release with a V1 kit?

drigz, 2018-04-23:
@juggledad What difference in filenames are you referring to? For the Assistant you need to create assistant.json and for Cloud Speech (& Check Cloud) you need to create cloud_speech.json.

Thanks for pointing out the voice-v1 mistake - we'll fix that in the next update.

The "Forbidden for url" error looks like #321 - if you [re-enable the Assistant API](https://console.cloud.google.com/apis/library/embeddedassistant.googleapis.com/) does it help?

juggledad, 2018-04-23:
> @juggledad What difference in filenames are you referring to? For the Assistant you need to create assistant.json and for Cloud Speech (& Check Cloud) you need to create cloud_speech.json.

Why would you have two different names for the credientials file?

Ok, re-enabling the Assistant API got it to work

drigz, 2018-04-23:
> Why would you have two different names for the credientials file?

They use different kinds of credentials. The Cloud Speech API uses a Service Account (cloud_speech.json contains a private key for the Service Account) whereas the Assistant needs access to your user account (assistant.json contains the "client secrets" used to start the process of getting partial access to your user account). Sorry if that's a confusing explanation - the authentication side of things is quite complex.

commit a7e18803 (2018-08-07): Fix broken link in Check Cloud

Fixes #336.

Change-Id: I010bc247a8bcbb1ad7f29d5ce2642b5815755c0f
closed 2018-06-21 by dmitriykovalev