google/aiyprojects-raspbian #337: Vision AIY Kit Not Booting

navneetmenon, 2018-04-22:
Hi
When I first assembled the kit and turned it on, everything was working fine. Joy Detection demo was working. After that, I tried connecting the kit to the android app and my computer to change the demo. Since then the camera is not loading, privacy led not turning on and there is no sound after boot up. 

The error in dmesg is **failed to bind reset GPIO**. Can anyone help with this? 

Here is the complete dmesg output - 
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 Sun Oct 29 11:47:10 GMT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0914e10, node_mem_map d7c8b900
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:EC:B0:AB vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=313dae9a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370980K/393216K available (5950K kernel code, 491K rwdata, 1948K rodata, 396K init, 725K bss, 14044K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05d7a48   (5951 kB)
                     .init : 0xc0841000 - 0xc08a4000   ( 396 kB)
                     .data : 0xc08a4000 - 0xc091ef48   ( 492 kB)
                      .bss : 0xc091ef48 - 0xc09d4648   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000062] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000151] bcm2835: system timer (irq = 27)
[    0.000628] Console: colour dummy device 80x30
[    0.000653] console [tty1] enabled
[    0.000679] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060311] pid_max: default: 32768 minimum: 301
[    0.060740] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060756] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062003] Disabling memory control group subsystem
[    0.062139] CPU: Testing write buffer coherency: ok
[    0.062196] ftrace: allocating 21715 entries in 64 pages
[    0.180407] Setting up static identity map for 0x8200 - 0x8238
[    0.182385] devtmpfs: initialized
[    0.192699] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.193123] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.193150] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.194432] pinctrl core: initialized pinctrl subsystem
[    0.195864] NET: Registered protocol family 16
[    0.198404] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.208113] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.208129] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.208231] Serial: AMBA PL011 UART driver
[    0.211143] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.211870] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.262763] bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
[    0.265431] SCSI subsystem initialized
[    0.265677] usbcore: registered new interface driver usbfs
[    0.265788] usbcore: registered new interface driver hub
[    0.265983] usbcore: registered new device driver usb
[    0.270449] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-08-08 12:05
[... 152 lines cut ...]
[   22.580949] Bluetooth: RFCOMM socket layer initialized
[   22.580978] Bluetooth: RFCOMM ver 1.11
[   23.376611] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[   24.953408] ktd202x 1-0030: Driver loaded for a ktd2026.
[   24.954943] aiy-vision spi0.0: Initializing
[   24.972648] aiy-vision spi0.0: Failed to bind reset GPIO
[   25.286527] aiy-io-i2c 1-0051: Setting board type vision
[   25.287224] aiy-io-i2c 1-0051: Driver loaded
[   25.288677] aiy-vision spi0.0: Initializing
[   25.310382] aiy-vision spi0.0: Failed to bind reset GPIO
[   28.463226] pwm-aiy-io pwm-aiy-io: Driver loaded
[   28.464698] aiy-vision spi0.0: Initializing
[   28.479162] aiy-vision spi0.0: Resetting myriad on probe
[   28.479179] aiy-vision spi0.0: Resetting myriad
[   28.559200] gpio-aiy-io gpio-aiy-io: Driver loaded
[   29.046059] aiy-adc aiy-adc: Vision bonnet ADC configuration.
[   29.273180] aiy-adc aiy-adc: Driver loaded
[   29.993882] systemd[1]: Found device /dev/ttyGS0.
[   29.999503] systemd[1]: apt-daily.timer: Adding 2h 29min 23.014283s random time.
[   30.074137] systemd[1]: apt-daily-upgrade.timer: Adding 58min 38.567469s random time.
[   31.209098] aiy-vision spi0.0: Writing myriad firmware
[   32.171965] random: crng init done
[   35.570975] aiy-vision spi0.0: Myriad booting
[   35.798102] aiy-vision spi0.0: Myriad ready
[   50.297382] fuse init (API version 7.26)

phypix, 2018-04-22:
I have a similar issue from the beginning. Following are issues and resolutions I tried, but cannot get the vision kit working.

1) The Raspberry pi zero does not bootup
Used: https://aiyprojects.withgoogle.com/help#updating-your-sd-card
Resolved by reflash SD Card with latest image - aiyprojects-2018-04-13.img.xz

2) Vision kit boots-up green privacy light lights up along with beep sound, but nothing happens.
- I tired see console output by attaching a miniHdmi cable to external monitor. After the raspberry pi boots up it just hangs at the black screen. Over couple of boot-ups I do see the raspberry pi boot-up with a desktop , but if I try to open terminal, the system hangs.

I wanted to try the commands from https://aiyprojects.withgoogle.com/help#troubleshooting-tips
but since the pi does not boot-up consistently and just hangs i'm stuck.. https://www.youtube.com/watch?v=l0rAr3XQcNM

ErnieBoxer, 2018-04-22:
I found if I ran the system updates it broke vision so i reflashed did not update and is working fine now. Hope this helps.

ErnieBoxer, 2018-04-22:
FYI: make sure you have proper power supply as usb from computer is said to not have enough juice. I have powered mine from a 3.0 port with no issues however.

phypix, 2018-04-22:
I was able boot-up the pi and got the following from dmesg. 
-------------------------
Full output 
-------------------------
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.80+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1098 Fri Mar 9 18:51:28 GMT 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c092b218, node_mem_map d7c8b800
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:52:01:07 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=fa081ec0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370892K/393216K available (6005K kernel code, 492K rwdata, 1968K rodata, 404K init, 725K bss, 14132K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05e5970   (6007 kB)
                     .init : 0xc0855000 - 0xc08ba000   ( 404 kB)
                     .data : 0xc08ba000 - 0xc0935368   ( 493 kB)
                      .bss : 0xc0935368 - 0xc09eaa88   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000027] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000068] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000167] bcm2835: system timer (irq = 27)
[    0.000657] Console: colour dummy device 80x30
[    0.001424] console [tty1] enabled
[    0.001479] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060345] pid_max: default: 32768 minimum: 301
[    0.060809] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060880] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062147] Disabling memory control group subsystem
[    0.062353] CPU: Testing write buffer coherency: ok
[    0.062458] ftrace: allocating 21948 entries in 65 pages
[    0.181089] Setting up static identity map for 0x8200 - 0x8238
[    0.183145] devtmpfs: initialized
[    0.192705] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.193208] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.193308] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.194620] pinctrl core: initialized pinctrl subsystem
[    0.196111] NET: Registered protocol family 16
[    0.198680] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.208429] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.208523] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.208665] Serial: AMBA PL011 UART driver
[    0.211584] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.212383] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.260856] bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
[    0.263660] SCSI subsystem initialized
[    0.264190] usbcore: registered new interface driver usbfs
[    0.264353] usbcore: registered new interface driver hub
[    0.264585] usbcore: registered new device driver usb
[    0.271166] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-03-13 18:52
[    0.273149] clocksource: Switched to clocksource timer
[    0.326885] VFS: Disk quotas dquot_6.6.0
[    0.327084] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[... 222 lines cut ...]
[  109.395573] 1f40: 00000000 d70049e0 d70049c0 c0042ac4 00000000 00000000 d7101fac d7101f68
[  109.395588] 1f60: c003ea54 c0042ad0 aaaafaaa 00000001 00000000 d70049c0 00000000 d7101f7c
[  109.395604] 1f80: d7101f7c 00000000 d7101f88 d7101f88 d70049e0 c003e95c 00000000 00000000
[  109.395622] 1fa0: 00000000 d7101fb0 c000fed4 c003e968 00000000 00000000 00000000 00000000
[  109.395643] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... cut ...]
[  119.201273] NOHZ: local_softirq_pending 40
[  138.250065] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  138.250142] Bluetooth: HIDP socket layer initialized
[  138.368810] hid-generic 0005:046D:B316.0002: unknown main item tag 0x0
[  138.393544] input: Logitech K760 as /devices/platform/soc/20201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:046D:B316.0002/input/input1
[  138.401467] hid-generic 0005:046D:B316.0002: input,hidraw1: BLUETOOTH HID v1.1b Keyboard [Logitech K760] on b8:27:eb:f8:ab:ad

phypix, 2018-04-25:
I contacted Google AIY Support which suggested following Debug Steps sharing here if you are having similar boot problems, in conclusion the memory dump in the dmesg translates to a faulty vision-bonnet hardware issue.
------------
Thank you for your interest into AIY projects.
Can you please reverify that the flex cables are connected properly and latched. 

Additionally, can you please perform following steps:
1. Can you please unplug the camera flex and try to booth the pi ? (With bonnet connected)
2. Run this command then : i2cdetect -y 1 (Share the output if you are able to see a terminal)
3. dmesg | grep myriad (Share the output)
4. sudo systemctl stop joy_detection_demo.service
5. raspistill -o file.jpg (Share the error)
6. leds_example.py (Is your button flashing now ? Close by pressing ctrl + c)
7. buzzer_demo.py (Did the device make any noice ? Close by pressing ctrl + c)
Then, Connect the camera flex directly on the Pi board (Remove Short flex from Pi)
8. raspistill -o ImageWithoutBonnet.jpg (Share what happens)
9. lsmod (Share the output)
-----------------------
From the above steps we concluded that the Raspberry pi and camera module worked fine in mycase. Camera when connected to vision-bonnet is when I'm having the boot issues or unresponsiveness.
-----------------------
Look for following Memory dump in dmesg -
[  125.256372] ------------[ cut here ]------------
[  125.256417] kernel BUG at kernel/smpboot.c:136!
[  125.256429] Internal error: Oops - BUG: 0 [#1] ARM
[  125.256441] Modules linked in: fuse aiy_adc(O) gpio_aiy_io(O) industrialio pwm_aiy_io(O) cmac rfcomm leds_ktd202x(O) bnep aiy_io_i2c(O) aiy_vision(O) spidev hci_uart btbcm bluetooth usb_f_rndis u_ether usb_f_acm u_serial brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio fixed pwm_soft(O) i2c_dev libcomposite dwc2 udc_core ip_tables x_tables ipv6
[  125.256577] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G           O    4.9.80+ #1098
[  125.256593] Hardware name: BCM2835
[  125.256605] task: d70a9b40 task.stack: d7100000
[  125.256632] PC is at smpboot_thread_fn+0x11c/0x178
[  125.256663] LR is at run_ksoftirqd+0x3c/0x58
[  125.256675] pc : [<c0042be0>]    lr : [<c0025fa0>]    psr: 60000013
               sp : d7101f40  ip : d7101f30  fp : d7101f64
[  125.256687] r10: 00000000  r9 : 00000002  r8 : c08c3efc
[  125.256697] r7 : 00000001  r6 : 00000000  r5 : d7100000  r4 : d70049c0
[  125.256708] r3 : d7100000  r2 : 00000000  r1 : d70a9b40  r0 : d7100001
[  125.256722] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  125.256732] Control: 00c5387d  Table: 11f4c008  DAC: 00000055
[  125.256743] Process ksoftirqd/0 (pid: 3, stack limit = 0xd7100188)
[  125.256754] Stack: (0xd7101f40 to 0xd7102000)
[  125.256770] 1f40: 00000000 d70049e0 d70049c0 c0042ac4 00000000 00000000 d7101fac d7101f68
[  125.256788] 1f60: c003ea54 c0042ad0 bfaefbab 00000001 00000000 d70049c0 00000000 d7101f7c
[  125.256804] 1f80: d7101f7c 00000000 d7101f88 d7101f88 d70049e0 c003e95c 00000000 00000000
[  125.256819] 1fa0: 00000000 d7101fb0 c000fed4 c003e968 00000000 00000000 00000000 00000000
[  125.256835] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  125.256854] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 aaaaabaf aeafaaab
[  125.256899] [<c0042be0>] (smpboot_thread_fn) from [<c003ea54>] (kthread+0xf8/0x114)
[  125.256934] [<c003ea54>] (kthread) from [<c000fed4>] (ret_from_fork+0x14/0x20)
[  125.256954] Code: e5983010 e5940000 e12fff33 eaffffc4 (e7f001f2)
[  125.256973] ---[ end trace a29a9859cd52b5d7 ]---
--------------------------------------------------------

dmitriykovalev, 2018-04-26:
@navneetmenon 

You `dmesg` output looks good because it has `Myriad ready` message. Please ignore `failed to bind reset GPIO` message (that's confusing but that's not a real error, we'll fix this soon). 

Can you please share JoyDemo application log by running `journalctl -u joy_detection_demo.service -b` ?

By the way, make sure to use wall power supply, power from USB port may be unreliable.

dmitriykovalev, 2018-04-29:
@navneetmenon 

Thank you for sharing JoyDemo log, that's pretty common error message if camera is not connected properly. Could you please verify that both flex cables (short and long) are installed correctly? It also could be a loose flex cable connector (make sure all of them are properly locked and cable is fully inserted in the connector).  Are you sure that JoyDemo was working fine when you assembled it and ran for the first time?

navneetmenon, 2018-04-29:
I'll check the cables once more.

But yeah the first time I assembled it, it was working fine.

dmitriykovalev, 2018-05-10:
@navneetmenon Did you have chance to check camera connections? Is JoyDemo working now for you?

ajaymdesai, 2018-05-26:
I'm having similar issues. the camera works via raspstill but if I run any example or the joy_detector my system locks up. 

May 26 06:00:29 aiy-vision kernel: [   53.546814] ------------[ cut here ]------------
May 26 06:00:29 aiy-vision kernel: [   53.546976] kernel BUG at kernel/smpboot.c:136!
May 26 06:00:29 aiy-vision kernel: [   53.547092] Internal error: Oops - BUG: 0 [#1] ARM
May 26 06:00:29 aiy-vision kernel: [   53.547211] Modules linked in: aiy_adc(O) industrialio pwm_aiy_io(O) gpio_aiy_io(O) aiy_vision(O) leds_ktd202x(O) aiy_io_i2c(O) cmac spidev rfcomm bnep hci_uart btbcm bluetooth usb_f_acm u_serial brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem fixed uio_pdrv_genirq uio pwm_soft(O) i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
May 26 06:00:29 aiy-vision kernel: [   53.548449] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G           O    4.9.80+ #1098
May 26 06:00:29 aiy-vision kernel: [   53.548618] Hardware name: BCM2835
May 26 06:00:29 aiy-vision kernel: [   53.548705] task: d70a9b40 task.stack: d7100000
May 26 06:00:29 aiy-vision kernel: [   53.548826] PC is at smpboot_thread_fn+0x11c/0x178
May 26 06:00:29 aiy-vision kernel: [   53.548950] LR is at run_ksoftirqd+0x3c/0x58
May 26 06:00:29 aiy-vision kernel: [   53.549054] pc : [<c0042be0>]    lr : [<c0025fa0>]    psr: 60000013
May 26 06:00:29 aiy-vision kernel: [   53.549054] sp : d7101f40  ip : d7101f30  fp : d7101f64
May 26 06:00:29 aiy-vision kernel: [   53.549291] r10: 00000000  r9 : 00000002  r8 : c08c3efc
May 26 06:00:29 aiy-vision kernel: [   53.549411] r7 : 00000001  r6 : 00000000  r5 : d7100000  r4 : d70049c0
May 26 06:00:29 aiy-vision kernel: [   53.549553] r3 : d7100000  r2 : 00000000  r1 : d70a9b40  r0 : d7100001
May 26 06:00:29 aiy-vision kernel: [   53.551628] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
May 26 06:00:29 aiy-vision kernel: [   53.553749] Control: 00c5387d  Table: 16ae4008  DAC: 00000055
May 26 06:00:29 aiy-vision kernel: [   53.555806] Process ksoftirqd/0 (pid: 3, stack limit = 0xd7100188)
May 26 06:00:29 aiy-vision kernel: [   53.557856] Stack: (0xd7101f40 to 0xd7102000)
May 26 06:00:29 aiy-vision kernel: [   53.559933] 1f40: 00000000 d70049e0 d70049c0 c0042ac4 00000000 00000000 d7101fac d7101f68
May 26 06:00:29 aiy-vision kernel: [   53.564052] 1f60: c003ea54 c0042ad0 aaaaaaaa 00000001 00000000 d70049c0 00000000 d7101f7c
May 26 06:00:29 aiy-vision kernel: [   53.568160] 1f80: d7101f7c 00000000 d7101f88 d7101f88 d70049e0 c003e95c 00000000 00000000
May 26 06:00:29 aiy-vision kernel: [   53.572242] 1fa0: 00000000 d7101fb0 c000fed4 c003e968 00000000 00000000 00000000 00000000
May 26 06:00:29 aiy-vision kernel: [   53.576416] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
May 26 06:00:29 aiy-vision kernel: [   53.580889] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 aaaaaeaa aaaaaaaa
May 26 06:00:29 aiy-vision kernel: [   53.585621] [<c0042be0>] (smpboot_thread_fn) from [<c003ea54>] (kthread+0xf8/0x114)
May 26 06:00:29 aiy-vision kernel: [   53.590469] [<c003ea54>] (kthread) from [<c000fed4>] (ret_from_fork+0x14/0x20)
May 26 06:00:29 aiy-vision kernel: [   53.593096] Code: e5983010 e5940000 e12fff33 eaffffc4 (e7f001f2) 
May 26 06:00:29 aiy-vision kernel: [   53.595759] ---[ end trace 2d4b1e19194d46e4 ]---

ajaymdesai, 2018-05-26:
pi@aiy-vision:~/AIY-projects-python/src/examples/vision $ ./image_classification_camera.py 

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.156187] Internal error: Oops: 80000005 [#1] ARM

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158173] Internal error: Oops: 5 [#2] ARM

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158360] Process python3 (pid: 1078, stack limit = 0xd7102188)

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158365] Stack: (0xd7103980 to 0xd7104000)

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158380] 3980: 00000a40 00002900 00000a40 dad7802c 00000010 00000080 0000000f 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158391] 39a0: d7229bb0 dad7802c 00000010 00000000 d7229bc0 d722aa18 00000000 d7002000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158401] 39c0: 00000001 000000ff d71039e4 d71039d8 c0366a48 c0365c30 d7103aac d71039e8

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158412] 39e0: c035b5f4 c0366a3c d7103a34 d7265000 d7103a28 00000000 00000007 d7002100

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158422] 3a00: ffffffff d72751a4 ffffffff 00000000 00000200 00000000 00000000 d7275000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158432] 3a20: d722a9f8 00000010 00000010 d7100001 d7262860 00000000 00000000 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158443] 3a40: 00000010 00000001 00000170 000000f0 00000080 00000010 00000007 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158454] 3a60: 00000001 d7229bb0 00000000 c0354e40 d7103a9c d7002000 d7275000 c0354e40

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158464] 3a80: d7103aac d7275000 d7002000 0000000f d722a9f8 00000010 c035b3a0 00000007

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158474] 3aa0: d7103aec d7103ab0 c0355164 c035b3ac 0000000f 0000002e 00000007 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158485] 3ac0: 0000003d d7002000 00000010 00000071 0000003e c09386c4 00000000 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158495] 3ae0: d7103b34 d7103af0 c0398660 c035507c 0000002e 36b4a597 d722a9f8 c09386c4

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158507] 3b00: d7002000 00000071 ffffffff c090cc8c 00000010 00000000 00000000 c09381a0

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158517] 3b20: c09381a0 00000004 d7103b5c d7103b38 c005b620 c0398310 c08bd760 60000193

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158528] 3b40: 00000125 00000010 60000193 c08c85ac d7103bb4 d7103b60 c005c628 c005b5a4

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158539] 3b60: 00000000 c08c85e0 00000000 00000000 d7103b9c d7103b80 c005695c c00bef30

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158549] 3b80: 00000000 00000010 c095aaee 00000010 c095aaee 00000008 60000193 c08c85ac

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158560] 3ba0: c09381a0 00000004 d7103c14 d7103bb8 c005cb50 c005c154 00000010 c08c85ac

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158571] 3bc0: c09381a0 00000000 d7103c34 d7103bd8 00000000 00000000 00000000 00000000

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158583] 3be0: c095a118 00000000 c095aaee bf0f5740 c08cbcbc d7103dc0 c072eae8 c08c1fc0

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158594] 3c00: 80000005 d51b8000 d7103c2c d7103c18 c005cf88 c005c7cc c0735fe8 d7103c4c

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158605] 3c20: d7103c44 d7103c30 c00e4a14 c005cf54 d7103c54 d7103c4c d7103c74 d7103c58

Message from syslogd@aiy-vision at May 26 06:25:37 ...
 kernel:[  501.158615] 3c40: c0084b98 c00e49f0 c0735fe8 bf0f574c d7103c58 00000000 00294f00 0000000b

dmitriykovalev, 2018-05-26:
@ajaymdesai There is a possible solution to kernel oopses you have, please take a look at latest comments in #346. Single-line solution from @dtreskunov is
```
echo over_voltage=4 | sudo tee -a /boot/config.txt && sudo shutdown -r now
```
Please let us know if that helps or not.

@navneetmenon most likely has a different problem related to loose flex cable connections.

ajaymdesai, 2018-05-29:
@dmitriykovalev your fix worked thanks the aiy kit now works.

closed 2018-06-21 by dmitriykovalev