google/aiyprojects-raspbian #338: Where do I find odler Images?

windigoHub, 2018-04-22:
The current image 13/4/18 doesn't work for me as described in #310 (stuck at the 4 raspberries screen with an underscore below them ticking indefinitely". 
But I used the voice kit before on 3/3/18, and the image I used then worked for me. Is there a way to find this or another older image? Or does someone still have it and can provide it?

drigz, 2018-04-23:
The previous image is available here: https://dl.google.com/dl/aiyprojects/aiyprojects-2018-02-21.img.xz

mikernorton, 2018-04-23:
I'm helping teach a class in using the AIY voice product, and would like to reduce variables by asking everyone to install the same image before the class. Is there a generic way to check when images are released and what the image names are? There are no tags or releases for newer images on github, and you can't view the list of images on the dl.google.com link, you have to know in the name in advance.

dmitriykovalev, 2018-05-05:
We are trying to make everything more organized, you can download old SD card images here https://github.com/google/aiyprojects-raspbian/releases

closed 2018-05-10 by dmitriykovalev