google/aiyprojects-raspbian #339: Check audio error: FileNotFoundError: [Errno 2] No such file or directory: '/proc/asound/cards'

voicehat, 2018-04-22:
Looking for help fixing this issue

dmitriykovalev, 2018-04-23:
Which audio board do you use, Voice Hat (for Pi 3) or Voice Bonnet (for Pi Zero)?

voicehat, 2018-04-24:
Voice Hat for pi 3

dmitriykovalev, 2018-04-24:
Can you please verify that you are using the latest SD card image from our [website](https://aiyprojects.withgoogle.com/)?  If you are still having this issue, can you manually type `sudo dtoverlay googlevoicehat-soundcard` in the terminal  and then run check audio script again?

dmitriykovalev, 2018-04-25:
@IchordeDionysos  Perfect! You can put `dtoverlay=googlevoicehat-soundcard` string into the `/boot/config.txt` file to load overlay on every Pi boot. We'll try to figure out how to improve our SD-card image. 

@voicehat Did this help to solve your problem?

blandiant, 2018-04-25:
hi, @dmitriykovalev 
i have the same problem, 
i wrote the command : `sudo dtoverlay googlevoicehat-soundcard`
and it work only once, I retried after I reboot of the raspberry pi and it doesn't work anymore, 
the terminal show this :
`Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 172, in <module>
    main()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 167, in main
    do_checks()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 129, in do_checks
    if not check_voicehat_present():
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 75, in check_voicehat_present
    return any(card_id in card for card in get_sound_cards().values())
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 50, in get_sound_cards
    with open(CARDS_PATH) as f:  # pylint: disable=invalid-name
FileNotFoundError: [Errno 2] No such file or directory: '/proc/asound/cards'
Press Enter to close...
`
thank you
ps: i tried to type the command in the /boot/config.txt, but i don't have the permission and i tried to add the permission in the properties of the file but I'm not allowed  to do that too

dmitriykovalev, 2018-04-26:
@blandiant 

That's pretty weird, `sudo dtoverlay googlevoicehat-soundcard` should always work. You have to run it every time after reboot.  Another way is to edit `/boot/config.txt` and insert (slightly different!) command `dtoverlay=googlevoicehat-soundcard` which does the same on every boot automatically. You need to use `sudo` in order to edit `/boot/config.txt` file.  More info about this configuration file and how to edit it — https://elinux.org/R-Pi_configuration_file

voicehat, 2018-04-28:
Typing dtoverlay=googlevoicehat-soundcard into / boot/ config.txt worked
for me too.

I think I used *sudo nano /boot/config.txt *to make the edit.

Thanks.

blandiant, 2018-05-04:
Thank you a lot, it works now,
I only have a small problem, when i use the "check audio", i say "1 2 3 testing" the sound volume of the recorded audio is low, even when i speak very close to the microphones
And when I use the assistant example I have to say "ok google" really close to the microphones to activate the assistant, otherwise it doesn't work
Do I open an new issue ?
Thank you again

drigz, 2018-05-04:
@blandiant Since the volume issue was mentioned in the comments of #307 as well, I've filed #356 to track it.