google/aiyprojects-raspbian #341: No information provided with release of a new image

LondonSi72, 2018-04-23:
@drigz, or any of the guys who are creating the code for this project, is there a published list of changes made with each new image?  It's very difficult as an end-user to figure out what has changed from one image to the next, and knowing whether it's worth updating from one image to another.

af987, 2018-04-23:
+1 on this.  

Some sort of release note, or at the very least an update of the issues for this GitHub project, would be great.  A simple update to close issues resolved in the latest image would be great.

My personal circumstance is that I'm on the previous (not the latest) image and have worked around a number of issues.  It would take at least half a day of my time to upgrade to the latest image and check whether the work arounds are still needed and get back to where I am now on the new image (assuming I didn't find something new that made me want to revert to the image I'm using today).  Whilst I think this project (and the wider Google AIY initiative) is great, I don't have half a day to burn every time a new image is released given that I have no way of knowing what I'm going to gain for the pain.  I'll even volunteer to pull a release note together if someone just pings me a list of issue #s that are resolved.

saket424, 2018-04-27:
@LondonSi72 
Do you mind sharing a link to the new image you are referring to? The link from the releases is dated Jan3
I assume you are referring to something newer

saket424, 2018-04-28:
The new image is at https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz

https://aiyprojects.withgoogle.com/help/#updating-your-sd-card

dmitriykovalev, 2018-08-16:
File `CHANGES.md` has been recently added.

closed 2018-08-16 by dmitriykovalev