google/aiyprojects-raspbian #342: SpicommTimeoutError

cygnus77, 2018-04-23:
Demo projects throw a SpicommTimeoutError (example below); sometimes after working for a little bit.
Thanks in advance for any troubleshooting suggestions !

Info:
- `dmesg` output: [dmesg2.txt](https://github.com/google/aiyprojects-raspbian/files/1940043/dmesg2.txt)
- `i2cdetect -y 1` output: [i2c.txt](https://github.com/google/aiyprojects-raspbian/files/1940044/i2c.txt)
- Use a 3A charger to power the system.
- Flashed with 4/13 image
- Notice "failed to bind reset GPIO" messages in dmesg output

`
pi@visionkit:~/AIY-projects-python/src/examples/vision $ ./image_classification_camera.py 
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 62] Timer expired

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./image_classification_camera.py", line 87, in <module>
    main()
  File "./image_classification_camera.py", line 76, in main
    with CameraInference(image_classification.model()) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 68, in __init__
    self._key = self._engine.load_model(descriptor)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 201, in load_model
    _check_firmware_info(self.get_firmware_info())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 275, in get_firmware_info
    info = self._communicate(request).firmware_info
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 138, in transact
    raise SpicommTimeoutError
aiy._drivers._spicomm.SpicommTimeoutError
`

dmitriykovalev, 2018-04-26:
Thanks for reporting this problem, we've never seen this before. Could you please provide more details? E.g. how long does it usually run before the crash, what happens if you restart it right after the crash, and etc.  Just ignore `failed to bind reset GPIO` message (not a real error) as longs as you have `Myriad ready` in the log.

cygnus77, 2018-04-28:
If the device was booted after being shutdown for a while, it works ok for some time. Then starts throwing spicomm timeouts. Even after reboots, it wouldn't work. I suspected a hardware issue and returned it to the store.

HeatfanJohn, 2018-05-09:
Do you happen to see the following errors in `dmesg` or `/var/log/syslog`?

```
[542083.837300] aiy-vision spi0.0: Slave not ready after 1000 ms
[542083.837323] aiy-vision spi0.0: Failed on SPI read
[542083.837354] aiy-vision spi0.0: Fatal error, resetting
[542083.837366] aiy-vision spi0.0: Resetting myriad
[542086.027442] aiy-vision spi0.0: Writing myriad firmware
[542092.466626] aiy-vision spi0.0: Myriad booting
[542092.647817] aiy-vision spi0.0: Myriad ready

```

I get this while running the `joy_detection_demo.service` every once in a while.  The Myriad reset causes an exception that doesn't get processed until you stop the service.  I added a logger display to the JoyDetector.run finally clause that gets executed as soon as the Myriad reset is logged.  The traceback stack doesn't get displayed until the service is stopped.

The Myriad reset appears to trigger an `OSError: [Errno 14] Bad address` exception in _spicomm.py.

I'm not an experience Python coder so I'm struggling to figure out how best to handle and ignore this exception and reset.

```
May  9 14:13:04 aiyvision kernel: [542083.837300] aiy-vision spi0.0: Slave not ready after 1000 ms
May  9 14:13:04 aiyvision kernel: [542083.837323] aiy-vision spi0.0: Failed on SPI read
May  9 14:13:04 aiyvision kernel: [542083.837354] aiy-vision spi0.0: Fatal error, resetting
May  9 14:13:04 aiyvision kernel: [542083.837366] aiy-vision spi0.0: Resetting myriad
May  9 14:13:06 aiyvision kernel: [542086.027442] aiy-vision spi0.0: Writing myriad firmware
May  9 14:13:13 aiyvision kernel: [542092.466626] aiy-vision spi0.0: Myriad booting
May  9 14:13:13 aiyvision kernel: [542092.647817] aiy-vision spi0.0: Myriad ready
**May  9 14:13:13 aiyvision python3[8898]: ERROR:__main__:JoyDetector: finnally called**

```
Traceback stack:

```
May  9 17:02:34 aiyvision systemd[1]: Stopping AIY Joy Detection Demo...
May  9 17:02:34 aiyvision python3[8898]: INFO:__main__:Stopping...
May  9 17:02:34 aiyvision python3[8898]: Traceback (most recent call last):
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
May  9 17:02:34 aiyvision python3[8898]:     fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
May  9 17:02:34 aiyvision python3[8898]: OSError: [Errno 14] Bad address
May  9 17:02:34 aiyvision python3[8898]: During handling of the above exception, another exception occurred:
May  9 17:02:34 aiyvision python3[8898]: Traceback (most recent call last):
May  9 17:02:34 aiyvision python3[8898]:   File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 276, in run
May  9 17:02:34 aiyvision python3[8898]:     for i, result in enumerate(inference.run()):
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
May  9 17:02:34 aiyvision python3[8898]:     yield self._engine.camera_inference()
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 257, in camera_inference
May  9 17:02:34 aiyvision python3[8898]:     return self._communicate(request).inference_result
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
May  9 17:02:34 aiyvision python3[8898]:     response.ParseFromString(self._transport.send(request.SerializeToString()))
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
May  9 17:02:34 aiyvision python3[8898]:     return self._spicomm.transact(request)
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 152, in transact
May  9 17:02:34 aiyvision python3[8898]:     raise SpicommInternalError
May  9 17:02:34 aiyvision python3[8898]: aiy._drivers._spicomm.SpicommInternalError
May  9 17:02:34 aiyvision python3[8898]: During handling of the above exception, another exception occurred:
May  9 17:02:34 aiyvision python3[8898]: Traceback (most recent call last):
May  9 17:02:34 aiyvision python3[8898]:   File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 291, in run
May  9 17:02:34 aiyvision python3[8898]:     break
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
May  9 17:02:34 aiyvision python3[8898]:     self.close()
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
May  9 17:02:34 aiyvision python3[8898]:     self._engine.stop_camera_inference()
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 263, in stop_camera_inference
May  9 17:02:34 aiyvision python3[8898]:     self._communicate(request)
May  9 17:02:34 aiyvision python3[8898]:   File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
May  9 17:02:34 aiyvision python3[8898]:     raise InferenceException(response.status.message)
May  9 17:02:34 aiyvision python3[8898]: aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.
May  9 17:02:34 aiyvision python3[8898]: During handling of the above exception, another exception occurred:
May  9 17:02:34 aiyvision python3[8898]: Traceback (most recent call last):
[... cut ...]
May  9 17:02:34 aiyvision python3[8898]: picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
May  9 17:02:34 aiyvision systemd[1]: joy_detection_demo.service: Main process exited, code=exited, status=1/FAILURE
May  9 17:02:34 aiyvision systemd[1]: Stopped AIY Joy Detection Demo.
May  9 17:02:34 aiyvision systemd[1]: joy_detection_demo.service: Unit entered failed state.
May  9 17:02:35 aiyvision systemd[1]: joy_detection_demo.service: Failed with result 'exit-code'.

```

kissste, 2018-06-07:
I have got the same issue. 

This is in dmesg:

``[ 2741.378777] aiy-vision spi0.0: Slave not ready after 1000 ms
[ 2741.378800] aiy-vision spi0.0: Failed on SPI read.
[ 2741.378830] aiy-vision spi0.0: Fatal error, resetting
[ 2741.378842] aiy-vision spi0.0: Resetting myriad
[ 2743.866702] aiy-vision spi0.0: Writing myriad firmware
[ 2749.881352] aiy-vision spi0.0: Myriad booting
[ 2750.062564] aiy-vision spi0.0: Myriad ready
[ 3100.102155] aiy-vision spi0.0: Slave not ready after 1000 ms
[ 3100.102171] aiy-vision spi0.0: Failed on SPI read.
[ 3100.102194] aiy-vision spi0.0: Fatal error, resetting
[ 3100.102202] aiy-vision spi0.0: Resetting myriad
[ 3102.580335] aiy-vision spi0.0: Writing myriad firmware
[ 3109.283709] aiy-vision spi0.0: Myriad booting
[ 3109.464890] aiy-vision spi0.0: Myriad ready
```

sainttelant, 2018-06-14:
yes, i got same errors, so what happened? any solutions?

lsx84, 2019-06-27:
I also got the same errors. After a lot of searching (and frustration) I stumbled upon [this](https://www.raspberrypi.org/forums/viewtopic.php?p=1276497) thread, to be precise [this](https://www.raspberrypi.org/forums/viewtopic.php?p=1276497#p1280058) post. It showed a line in the /boot/config.txt file which was commented in my config.txt:

`dtparam=spi=on`

Uncommented the line, rebooted and it now works like a charm!

I hope this will fix the error on your device. Otherwise that thread has a lot of things you could check/do.

dmitriykovalev, 2019-11-14:
If everything is configured properly then Pi should load device tree overlay from the Vision Bonnet board. In this case `/boot/config.txt` should have `dtparam=spi=on` line commented out (i.e. it's disabled).

closed 2019-11-14 by dmitriykovalev