google/aiyprojects-raspbian #346: Vision kit JOY sample stops working

saket424, 2018-04-27:
I have a new Vision Kit from Target that I assembled. After working initially for a few minutes, the vision kit JOY sample stops working with the following output in the dmesg log. I have the unit powered with a 5V 2.5 A powersupply. Any suggestions ?

[  252.549751] Unregistered device pwm22
[  317.266061] ------------[ cut here ]------------
[  317.266115] kernel BUG at mm/slub.c:3873!
[  317.266126] Internal error: Oops - BUG: 0 [#1] ARM
[  317.266138] Modules linked in: fuse aiy_adc(O) industrialio gpio_aiy_io(O) pwm_aiy_io(O) cmac rfcomm bnep aiy_io_i2c(O) leds_ktd202x(O) hci_uart btbcm bluetooth aiy_vision(O) spidev usb_f_rndis u_ether usb_f_acm u_serial brcmfmac brcmutil snd_soc_bcm2835_i2s(O) regmap_mmio cfg80211 snd_soc_core rfkill snd_compress snd_pcm_dmaengine snd_pcm snd_timer i2c_bcm2835 snd spi_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio fixed pwm_soft(O) i2c_dev libcomposite dwc2 udc_core ip_tables x_tables ipv6
[  317.266304] CPU: 0 PID: 259 Comm: python3 Tainted: G           O    4.9.59+ #1047
[  317.266315] Hardware name: BCM2835
[  317.266328] task: d6806d00 task.stack: cccfa000
[  317.266368] PC is at kfree+0x144/0x18c
[  317.266406] LR is at kvfree+0x54/0x5c
[  317.266419] pc : [<c0134f90>]    lr : [<c01071b4>]    psr: 40000013
               sp : cccfbe18  ip : cccfbe40  fp : cccfbe3c
[  317.266432] r10: 0000084b  r9 : d8826000  r8 : c00eddd0
[  317.266442] r7 : c00ee078  r6 : c01071b4  r5 : cccfbe50  r4 : cce13f9d
[  317.266452] r3 : d7e58c4c  r2 : 00000100  r1 : 00000100  r0 : cccfbe50
[  317.266468] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  317.266479] Control: 00c5387d  Table: 0cde4008  DAC: 00000055
[  317.266490] Process python3 (pid: 259, stack limit = 0xcccfa188)
[  317.266500] Stack: (0xcccfbe18 to 0xcccfc000)
[  317.266513] be00:                                                       00000001 c0914e10
[  317.266529] be20: cce13f80 cce13f9d cccfbe34 cccfbe38 cccfbe4c cccfbe40 c01071b4 c0134e58
[  317.266545] be40: cccfbe6c cccfbe50 c012514c c010716c cce13f80 00000000 000005d3 ae214018
[  317.266561] be60: cccfbe7c cccfbe70 c0125240 c01250c8 cccfbe94 cccfbe80 bf413c20 c01251f8
[  317.266576] be80: d509f400 00000000 cccfbefc cccfbe98 bf4154cc bf413bb0 00000000 d509f470
[  317.266591] bea0: d509f480 ae214008 00000001 00000018 00003a98 00c00010 0000007e 00000000
[  317.266608] bec0: d6806d00 c004ff58 cccfbec8 cccfbec8 00000000 ae214008 d6857190 d2ee4960
[  317.266623] bee0: 00000005 00000005 cccfa000 00000000 cccfbf7c cccfbf00 c01575f0 bf414fdc
[  317.266638] bf00: cccfbf24 c00703cc cccfbf4c cccfbf18 c00703cc c049d27c 80000000 00000009
[  317.266653] bf20: 00000189 00000000 00000001 00000081 003eb198 c016211c cd3d5480 00c00010
[  317.266671] bf40: ae214008 be9cdbd4 c0108903 00000005 cccfbf6c d2ee4961 ae214008 d2ee4960
[  317.266686] bf60: c0108903 00000005 cccfa000 00000000 cccfbfa4 cccfbf80 c0157d6c c0157560
[  317.266701] bf80: be9cdbd4 00c00010 00378000 be9cdbd4 00000036 c000ffc4 00000000 cccfbfa8
[  317.266717] bfa0: c000fe40 c0157d34 00c00010 00378000 00000005 c0108903 ae214008 be9cdbd4
[  317.266734] bfc0: 00c00010 00378000 be9cdbd4 00000036 00000005 c0108903 ae214008 be9cdbd8
[  317.266749] bfe0: 003782fc be9cdbc4 000362d4 b6d9f8ac 80000010 00000005 17ffa861 17ffac61
[  317.266789] [<c0134f90>] (kfree) from [<c01071b4>] (kvfree+0x54/0x5c)
[  317.266827] [<c01071b4>] (kvfree) from [<c012514c>] (__vunmap+0x90/0xe0)
[  317.266850] [<c012514c>] (__vunmap) from [<c0125240>] (vfree+0x54/0x94)
[  317.266897] [<c0125240>] (vfree) from [<bf413c20>] (transaction_unref+0x7c/0xa4 [aiy_vision])
[  317.266956] [<bf413c20>] (transaction_unref [aiy_vision]) from [<bf4154cc>] (visionbonnet_ioctl+0x4fc/0x6a8 [aiy_vision])
[  317.266994] [<bf4154cc>] (visionbonnet_ioctl [aiy_vision]) from [<c01575f0>] (do_vfs_ioctl+0x9c/0x7d4)
[  317.267014] [<c01575f0>] (do_vfs_ioctl) from [<c0157d6c>] (SyS_ioctl+0x44/0x6c)
[  317.267051] [<c0157d6c>] (SyS_ioctl) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[  317.267069] Code: 1a000003 e5932014 e3120001 1a000000 (e7f001f2) 
[  317.267085] ---[ end trace 8b8f08149c229173 ]---

bryantqo, 2018-04-29:
I had the same issue. Ran `sudo apt-get dist-upgrade` and testing again now.

saket424, 2018-05-01:
More logs when i run the face detection demo . The errors happen even with the 20180418 image

[   82.575807] Stopping timer.
[   82.628704] Unregistered device pwm22
[  130.673472] aiy-vision spi0.0: Transaction interrupted tid=1
[  186.537079] input: raspidisp-keyboard as /devices/virtual/input/input0
[  243.906471] ------------[ cut here ]------------
[  243.906528] WARNING: CPU: 0 PID: 1106 at drivers/staging/vc04_services/interface/vchiq_arm/vchiq_core.c:947 queue_message+0x464/0x998
[  243.906537] Modules linked in: evdev cmac rfcomm bnep aiy_adc(O) industrialio gpio_aiy_io(O) pwm_aiy_io(O) spidev leds_ktd202x(O) aiy_io_i2c(O) aiy_vision(O) hci_uart btbcm bluetooth usb_f_acm u_serial brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem fixed uio_pdrv_genirq uio uinput cuse fuse pwm_soft(O) i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[  243.906684] CPU: 0 PID: 1106 Comm: uv4l Tainted: G           O    4.9.80+ #1098
[  243.906689] Hardware name: BCM2835
[  243.906762] [<c0016068>] (unwind_backtrace) from [<c0013c90>] (show_stack+0x20/0x24)
[  243.906788] [<c0013c90>] (show_stack) from [<c031ba98>] (dump_stack+0x20/0x28)
[  243.906820] [<c031ba98>] (dump_stack) from [<c0021cb4>] (__warn+0xe4/0x10c)
[  243.906840] [<c0021cb4>] (__warn) from [<c0021da8>] (warn_slowpath_null+0x30/0x38)
[  243.906860] [<c0021da8>] (warn_slowpath_null) from [<c04cc190>] (queue_message+0x464/0x998)
[  243.906882] [<c04cc190>] (queue_message) from [<c04ce89c>] (vchiq_queue_message+0x108/0x140)
[  243.906902] [<c04ce89c>] (vchiq_queue_message) from [<c04d2d58>] (vchiq_ioctl+0x434/0x188c)
[  243.906933] [<c04d2d58>] (vchiq_ioctl) from [<c0157a58>] (do_vfs_ioctl+0x9c/0x7d4)
[  243.906951] [<c0157a58>] (do_vfs_ioctl) from [<c01581d4>] (SyS_ioctl+0x44/0x6c)
[  243.906973] [<c01581d4>] (SyS_ioctl) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[  243.906982] ---[ end trace aa2dcbe0c547b42f ]---
[  243.907041] Unable to handle kernel NULL pointer dereference at virtual address 00000015
[  243.907071] pgd = d40a8000
[  243.907080] [00000015] *pgd=14002831, *pte=00000000, *ppte=00000000
[  243.907112] Internal error: Oops: 817 [#1] ARM
[  243.907128] Modules linked in: evdev cmac rfcomm bnep aiy_adc(O) industrialio gpio_aiy_io(O) pwm_aiy_io(O) spidev leds_ktd202x(O) aiy_io_i2c(O) aiy_vision(O) hci_uart btbcm bluetooth usb_f_acm u_serial brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 spi_bcm2835 bcm2835_gpiomem fixed uio_pdrv_genirq uio uinput cuse fuse pwm_soft(O) i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[  243.907258] CPU: 0 PID: 1106 Comm: uv4l Tainted: G        W  O    4.9.80+ #1098
[  243.907272] Hardware name: BCM2835
[  243.907286] task: d5384420 task.stack: ce052000
[  243.907309] PC is at remote_event_signal+0x20/0x4c
[  243.907325] LR is at queue_message+0x190/0x998
[  243.907339] pc : [<c04d5a94>]    lr : [<c04cbebc>]    psr: 20000013
               sp : ce053d78  ip : ce053d88  fp : ce053d84
[  243.907353] r10: 00000010  r9 : c09c7b68  r8 : c0921524
[  243.907365] r7 : 00000002  r6 : d75bee00  r5 : ce053ea0  r4 : d5333e00
[  243.907375] r3 : 00000000  r2 : 00000001  r1 : 00000000  r0 : 00000011
[  243.907390] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  243.907403] Control: 00c5387d  Table: 140a8008  DAC: 00000055
[  243.907414] Process uv4l (pid: 1106, stack limit = 0xce052188)
[  243.907428] Stack: (0xce053d78 to 0xce054000)
[  243.907444] 3d60:                                                       ce053e04 ce053d88
[  243.907461] 3d80: c04cbebc c04d5a80 00000310 ffff0a11 00000000 00000200 00000200 c04d4a74
[  243.907478] 3da0: 00000006 00000003 00000000 d7580194 0500300a ce053e90 c0921524 00000005
[  243.907497] 3dc0: d5333e00 c09cbe38 fff9fee9 ffffffff 00005007 00000000 ce053e3c 0000000c
[  243.907514] 3de0: 00000002 00000010 d5333e00 ce053e90 d5333e00 00000000 ce053e3c ce053e08
[  243.907529] 3e00: c04ce89c c04cbd38 00000002 00000010 00000001 00000010 ce053e90 400cc404
[  243.907548] 3e20: c0921530 d6821000 00000010 00000000 ce053efc ce053e40 c04d2d58 c04ce7a0
[  243.907564] 3e40: ae4e5d1c 00000001 d401e828 ce053ea0 0000a000 00000000 0001d000 0001d000
[  243.907582] 3e60: 00000000 00000002 00000018 d401e828 00000000 d6bbe190 00000000 00000000
[  243.907599] 3e80: ce053f3c 00004003 00000002 ae4e5cb8 ae4e5cac 00000004 ae4e5ce4 0000000c
[  243.907616] 3ea0: 00000001 00000000 00000000 c004dab4 000660d0 00000000 d5384420 c004db28
[  243.907633] 3ec0: c9b70b68 c004dd94 ce053f0c ce053ed8 00000000 ae4e5c84 d7325cd0 d401ed20
[  243.907652] 3ee0: 00000006 00000006 ce052000 00000000 ce053f7c ce053f00 c0157a58 c04d2930
[  243.907670] 3f00: ce053f2c c00705d4 ce053f4c ce053f18 c00705d4 c04a9e7c 567b9fee 00076d83
[  243.907687] 3f20: ffffffff ce053f88 ae4e5cc8 00000000 0000004e c0162584 d514f480 000000b8
[  243.907704] 3f40: ae4e5c84 ae4e5c84 400cc404 00000006 ce053f6c d401ed21 ae4e5c84 d401ed20
[  243.907721] 3f60: 400cc404 00000006 ce052000 00000000 ce053fa4 ce053f80 c01581d4 c01579c8
[  243.907737] 3f80: 00004003 000000b8 b6a65334 ae4e5c84 00000036 c000ffc4 00000000 ce053fa8
[  243.907754] 3fa0: c000fe40 c015819c 000000b8 b6a65334 00000006 400cc404 ae4e5c84 00004003
[  243.907771] 3fc0: 000000b8 b6a65334 ae4e5c84 00000036 b5382a98 becf9588 b4a02bb0 0018316c
[  243.907790] 3fe0: b6a65240 ae4e5c7c b6a5403c b6bad80c 60000010 00000006 400636e6 13000000
[... cut ...]
[  243.907850] [<c04cbebc>] (queue_message) from [<c04ce89c>] (vchiq_queue_message+0x108/0x140)
[  243.907874] [<c04ce89c>] (vchiq_queue_message) from [<c04d2d58>] (vchiq_ioctl+0x434/0x188c)
[  243.907904] [<c04d2d58>] (vchiq_ioctl) from [<c0157a58>] (do_vfs_ioctl+0x9c/0x7d4)
[  243.907927] [<c0157a58>] (do_vfs_ioctl) from [<c01581d4>] (SyS_ioctl+0x44/0x6c)
[  243.907957] [<c01581d4>] (SyS_ioctl) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[  243.907982] Code: e8bd4000 e3a03000 ee073f9a e3a02001 (e5802004) 
[  243.908082] ---[ end trace aa2dcbe0c547b430 ]---

cmrigney, 2018-05-05:
I have the same issue but haven’t had the chance to check dmesg.  I just bought it from Target today, and it seems to randomly freeze, although it usually works for a half a minute or so. Using a 2.1 amp supply. I hope this gets resolved soon. Stinks it’s not working after I just bought it.

cmrigney, 2018-05-05:
It just now froze in the midst of starting and pulsing the piezo, so the piezo beeped until I pulled the plug. Seems random.

cmrigney, 2018-05-05:
I’m starting to think it’s a hardware issue, because the longer I leave it off before starting it back up, the longer it works without freezing. I’m going to exchange it and see if that fixes it.

bryantqo, 2018-05-05:
I am seeing the same thing. Also leaving the back open after it's cooled
off makes it run longer. I am going to pull mine out of the box and put a
cpu fan on it

cmrigney, 2018-05-06:
Status update. I exchanged it at Target and the new one works flawlessly! It’s as if there are some that have an overheating issue. But the new one is fully enclosed and all, no mods, yet works fine. 

Shows me for always skipping the first item on the shelf for fear of buying a man handled product :D

saket424, 2018-05-06:
I bought 3 of them. 2 were faulty and I had to return it and the third one was fine. So it is definitely some marginal movidius hardware causing us grief.

cmrigney, 2018-05-06:
Hmm, that’s insightful. I wonder who’s in charge of production quality? Anyone from Google have any input?

PeterMalkin, 2018-05-07:
We are investigating the issue. But in re: production, the manufacturing line producing the units is the same for all kits. The chipsets from Intel Movidius have to pass the QA when they go off their manufacturing line. So I doubt Intel wouldnt catch a problem if there was one. On our production line we have manufacturing tests that cover all the aspects of the hardware. We test the boards before and after the components soldering, and have different set of tests for every part of the device functionality.

cmrigney, 2018-05-07:
Thanks for the insight @PeterMalkin.  I'm guessing you guys test the SD cards as well?  I know we've had an issue with faulty SD cards at a company I worked at previously.  We also had issues where our product worked well in production testing, but then it would fail when the customer got it (it turned out to be a voltage spike randomly frying the board upon plugging it in).  Not suggesting anything like that here.  All that was to say, I have empathy for the grief this is probably causing you guys.

PeterMalkin, 2018-05-08:
Hey,

we have hard time reproducing this on our end reliably. If any of you happen to see this again - would you be willing to mail the kit over? I will mail back a new kit and a $20 gift card to compensate for shipping. That would really help us to get to the bottom of this.

Thanks!!

lleonid, 2018-05-08:
@saket424, @bryantqo what's the exact set up when you're running into those issues?
In particular:
Is display connected to mini hdmi port on Pi?
Is Pi connected to wifi and/or connected via micro usb data port?
What does your camera face? I.e. your face, monitor, wall?
Did you attempt to disassemble the kit? It's quite easy to damage vision bonnet when removing white standoffs from it.
If that happens again could you run 'vcgencmd measure_temp' on Pi and report back the output please?

@bryantqo, do you also have Pi connected to 2.1+A usb power supply?

bryantqo, 2018-05-08:
Yes power supply is sufficient. I will attempt to run the command however I
think the last time I ran it I ended up with a kernel panic which flat
lined the system. If I ssh into the board and stop the joy demo before the
crash the os keeps running.

HDMI: No
WiFi: Yes
Power: power micro USB
Disassembled: No
Facing the desk

ScottBriening, 2018-05-16:
I am experiencing the same thing. @PeterMalkin, I am willing to mail you my vision kit rather than exchange it at Target. (For Science. And for the community.) Overheating seems to be a likely culprit. I was debating on slapping heat sink on the zero to see if that would fix the problem, but will hold off until I hear back from you. In the meantime I'll work with the Voice Kit. Would love to try the Android Things setup. And DialogueFlow and Firebase and sill hafta try gVisor... So much to do, so little time!

PeterMalkin, 2018-05-16:
@ScottBriening 
It would be extremely kind of you to mail your kit here. I will mail you a new one back. Would you be so kind as to shoot me an email at petermalkin@google.com, and we coordinate the exchange?
Thank you!

dtreskunov, 2018-05-24:
I'm also having an issue with my Vision Kit where very shortly after starting the joy detection demo my ssh session becomes unresponsive. It seems related to the board getting too hot (which board?):
```
pi@raspberrypi:~ $ while true; do vcgencmd measure_temp;  sleep 5; done;
temp=43.3'C
temp=43.3'C
temp=44.4'C
temp=43.9'C
temp=43.3'C
temp=43.3'C
temp=44.9'C <- start joy detector demo
temp=46.5'C
temp=49.2'C
temp=49.8'C
temp=48.7'C
temp=49.2'C
temp=50.3'C
temp=51.4'C
temp=50.8'C <- chime sounds
temp=49.8'C
temp=51.4'C
temp=50.8'C
temp=53.0'C <- last reading printed to the console before I have to cut the power
```

I should note that I was seeing `oops` errors in dmesg output (and the demo stopped working but the OS seemed kind of OK-ish) before I switched the power supply from 1.5A to 2.1A. After this change, I can't even check `dmesg` output because nothing works anymore.

dmitriykovalev, 2018-05-24:
@dtreskunov What kind of `oops` have you seen? Could you attach HDMI monitor to get the exact error message?

You max temperature (53.0 C) is actually pretty low. In my experiments VisionKit was working fine when Pi was at 90.0 C. Everything below 85.0 C is considered normal I guess.

dtreskunov, 2018-05-25:
@dmitriykovalev Here's the `dmesg` output after `joy_detection_demo.py` stops working (shortly after being started): 

```
[  296.624859] Starting timer (pulse).
[  296.625076] Starting timer (period).
[  296.625291] Starting timer (pulse).
[  296.625523] Starting timer (period).
[  296.701112] Stopping timer.
[  296.719119] Unregistered device pwm22
[  296.804532] Unable to handle kernel NULL pointer dereference at virtual address 00000072
[  296.804578] pgd = d0100000
[  296.804589] [00000072] *pgd=101cc831, *pte=00000000, *ppte=00000000
[  296.804624] Internal error: Oops: 17 [#1] ARM
[  296.804638] Modules linked in: aiy_adc(O) pwm_aiy_io(O) gpio_aiy_io(O) industrialio cmac rfcomm bnep leds_ktd202x(O) aiy_io_i2c(O) hci_uart btbcm serdev bluetooth ecdh_generic aiy_vision(O) spidev usb_f_rndis u_ether usb_f_acm u_serial brcmfmac brcmutil cfg80211 rfkill snd_soc_bcm2835_i2s regmap_mmio i2c_bcm2835 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer spi_bcm2835 snd uio_pdrv_genirq fixed uio uinput cuse fuse pwm_soft(O) i2c_dev libcomposite dwc2 udc_core ip_tables x_tables ipv6
[  296.804823] CPU: 0 PID: 1040 Comm: python3 Tainted: G           O    4.14.34+ #1110
[  296.804833] Hardware name: BCM2835
[  296.804845] task: cf72e120 task.stack: d01c6000
[  296.804894] PC is at free_pcppages_bulk+0x274/0x498
[  296.804910] LR is at 0xd7ca70b4
[  296.804919] pc : [<c00fb068>]    lr : [<d7ca70b4>]    psr: 60000093
[  296.804930] sp : d01c7db8  ip : d7ca70a0  fp : d01c7e04
[  296.804938] r10: 00000000  r9 : d7ca710c  r8 : 00000001
[  296.804950] r7 : ffffffff  r6 : 00000ce8  r5 : c09be3d0  r4 : c09be368
[  296.804962] r3 : 00000034  r2 : c09be368  r1 : 00000002  r0 : d7cb7494
[  296.804978] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  296.804988] Control: 00c5387d  Table: 10100008  DAC: 00000055
[  296.805000] Process python3 (pid: 1040, stack limit = 0xd01c6188)
[  296.805011] Stack: (0xd01c7db8 to 0xd01c8000)
[  296.805024] 7da0:                                                       00000000 00000001
[  296.805039] 7dc0: d7ca710c c09be368 d7c72af8 d7c72aec 0000001c 0000001c d881f000 60000013
[  296.805054] 7de0: d7c72aec 0000001f c09be368 00000000 00000000 d7cad2a4 d01c7e44 d01c7e08
[  296.805070] 7e00: c00fd0ac c00fae00 d01c7e2c d01c7e18 c0136388 c09be368 00000002 00000000
[  296.805085] 7e20: d514e680 dbf89000 00000001 00000018 d010a4c0 d010a4dc d01c7e5c d01c7e48
[  296.805100] 7e40: c00fd13c c00fce94 d881f000 00000565 d01c7e7c d01c7e60 c0137bdc c00fd0f8
[  296.805116] 7e60: d010a4c0 00000000 000001eb ae214018 d01c7e8c d01c7e80 c0137cfc c0137b84
[  296.805131] 7e80: d01c7ea4 d01c7e90 bf433ba8 c0137cb4 d6ae6b00 00000000 d01c7f0c d01c7ea8
[  296.805147] 7ea0: bf4354cc bf433b38 d01d8de0 d6ae6b70 d6ae6b80 ae214008 c004c360 00000018
[  296.805163] 7ec0: 00001388 00c00010 000000a3 00000000 cf72e120 c00531d4 d01c7ed8 d01c7ed8
[  296.805180] 7ee0: c0046aec ae214008 d39a5c10 d01f55a0 00000006 00000006 d01c6000 00000000
[  296.805196] 7f00: d01c7f7c d01c7f10 c016c858 bf434fdc d01c7f74 d01c7f20 c06455ac c0046a9c
[  296.805210] 7f20: 00000001 00400000 fffffeb8 0000000a 1554f240 c01774c4 d0187600 00c00010
[  296.805226] 7f40: ae214008 bebdce54 c0108903 00000006 d01c7f6c d01f55a1 ae214008 d01f55a0
[  296.805241] 7f60: c0108903 00000006 d01c6000 00000000 d01c7fa4 d01c7f80 c016cf54 c016c7c8
[  296.805256] 7f80: bebdce54 00c00010 00378000 bebdce54 00000036 c000ff64 00000000 d01c7fa8
[  296.805273] 7fa0: c000fdc0 c016cf1c 00c00010 00378000 00000006 c0108903 ae214008 bebdce54
[  296.805289] 7fc0: 00c00010 00378000 bebdce54 00000036 00000006 c0108903 ae214008 bebdce58
[  296.805305] 7fe0: 003782fc bebdce44 000362d4 b6df080c 80000010 00000006 00000000 00000000
[  296.805349] [<c00fb068>] (free_pcppages_bulk) from [<c00fd0ac>] (free_hot_cold_page+0x224/0x264)
[  296.805371] [<c00fd0ac>] (free_hot_cold_page) from [<c00fd13c>] (__free_pages+0x50/0x54)
[  296.805402] [<c00fd13c>] (__free_pages) from [<c0137bdc>] (__vunmap+0x64/0xe0)
[  296.805422] [<c0137bdc>] (__vunmap) from [<c0137cfc>] (vfree+0x54/0x94)
[  296.805468] [<c0137cfc>] (vfree) from [<bf433ba8>] (transaction_unref+0x7c/0xa4 [aiy_vision])
[  296.805528] [<bf433ba8>] (transaction_unref [aiy_vision]) from [<bf4354cc>] (visionbonnet_ioctl+0x4fc/0x6a8 [aiy_vision])
[  296.805559] [<bf4354cc>] (visionbonnet_ioctl [aiy_vision]) from [<c016c858>] (do_vfs_ioctl+0x9c/0x754)
[  296.805580] [<c016c858>] (do_vfs_ioctl) from [<c016cf54>] (SyS_ioctl+0x44/0x6c)
[  296.805611] [<c016cf54>] (SyS_ioctl) from [<c000fdc0>] (ret_fast_syscall+0x0/0x28)
[  296.805632] Code: e58c0014 e7842003 e3a03034 e51b2040 (e0212193)
[  296.805650] ---[ end trace ee6907230b405e54 ]---
```

dmitriykovalev, 2018-05-26:
@dtreskunov  Could you try to add 
```
over_voltage=4
over_voltage_min=4
```
at the beginning of `/boot/config.txt`? I'm still checking but looks like it helps to eliminate similar problem. More context is here https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=212777

dtreskunov, 2018-05-26:
```echo over_voltage=4 | sudo tee -a /boot/config.txt && sudo shutdown -r now```

@dmitriykovalev Thanks so much! You just saved me a trip to the store. The problem seems to be gone after adding `over_voltage=4` to `/boot/config.txt`.

At idle, the CPU clock is 700MHz and voltage is 1.2V. While`joy_detection_demo.py` is running, the clock goes up to 1GHz and voltage to 1.4V. Before the `/boot/config.txt` change, the voltage would not increase when CPU was throttled up. For the record, this Raspberry Pi is reporting temps around 65'C while the demo is running.

For my own reference, here's [some documentation](https://elinux.org/RPI_vcgencmd_usage) on how to take these measurements using the `vcgencmd` utility.

dmitriykovalev, 2018-05-26:
@dtreskunov Great, I'm glad this solution helped! @ScottBriening, thank you very much for sending your board to us for experiments, it has the same issue and the same fix removed the problem.  We still need time to understand why :)

It's interesting that you didn't use `over_voltage_min` which affects dynamic frequency clocking, https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

ScottBriening, 2018-05-26:
@dmitriykovalev @PeterMalkin YAY!!! I just sent you an email suggesting a similar solution, but it looks like you beat me to the punch. Nice work guys. Now let's make some mAgIc!

ScottBriening, 2018-05-30:
@dmitriykovalev This is interesting indeed. So it's just the camera with pi0 combo that is the problem. I'm surprised this hasn't popped up more often / addressed by the folks at Pi.

From the raspberrypi.org docs, the default over_voltage setting for the pi zero is 6 (1.35V) and over_voltage_min is 0. By setting over_voltage to 4 (1.3V) it _should_ decrease the voltage supplied by 0.05V. Setting over_voltage_min to 4 _should_ ensure the voltage supplied goes no lower than 1.3 when the system is idle. It's great that these settings allow cameras to work with pi0, but... Why? (As it says in the link you provided "I don't know what to believe".)

Would love to tinker. Frying a couple pi0s isn't a big deal, but I certainly don't wanna lose the vision hat! Oh and you're welcome, of course. Happy to help! Got the new kit put together and so far so good. Oh, save the buzzer. Either I don't have enough joy or it's a dud. I'll get around to replacing it sometime.

ajaymdesai, 2018-05-30:
I had this problem but it's fixed based on the over_voltage fix. Thanks.

phillipeloher, 2018-06-12:
Same issue here but not fixed with over_voltage.  Returning soon.

dmitriykovalev, 2018-06-13:
@phillipeloher If possible, can you provide any additional details about your crash? e.g. dmesg output

d1981, 2018-06-27:
I'm getting the same issue if I allow the joy detection demo to run after bootup. I have HDMI connected to a monitor and mini unpowered usb hub to connect my logitech wireless keyboard/trackpad (1 usb dongle connected). 

Same issue after applying the voltage fix to /boot/config.txt. 

dmesg output with kernel error right before it locked up below. I bought this from target about a month ago and just got around to playing with it today.

Running apt-get upgrade to see if that helps.

`pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 Sun Oct 29 11:47:10 GMT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0914e10, node_mem_map d7c8b900
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:E3:F2:6A vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=86b6fce0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370980K/393216K available (5950K kernel code, 491K rwdata, 1948K rodata, 396K init, 725K bss, 14044K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05d7a48   (5951 kB)
                     .init : 0xc0841000 - 0xc08a4000   ( 396 kB)
                     .data : 0xc08a4000 - 0xc091ef48   ( 492 kB)
                      .bss : 0xc091ef48 - 0xc09d4648   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000063] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000155] bcm2835: system timer (irq = 27)
[    0.000631] Console: colour dummy device 80x30
[    0.000656] console [tty1] enabled
[    0.000681] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060313] pid_max: default: 32768 minimum: 301
[    0.060740] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060756] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062002] Disabling memory control group subsystem
[    0.062138] CPU: Testing write buffer coherency: ok
[    0.062195] ftrace: allocating 21715 entries in 64 pages
[    0.180401] Setting up static identity map for 0x8200 - 0x8238
[    0.182376] devtmpfs: initialized
[    0.192676] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.193104] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.193131] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.194409] pinctrl core: initialized pinctrl subsystem
[    0.195840] NET: Registered protocol family 16
[    0.198387] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.208081] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.208098] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.208200] Serial: AMBA PL011 UART driver
[    0.211124] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.211851] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.262757] bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
[    0.265428] SCSI subsystem initialized
[    0.265674] usbcore: registered new interface driver usbfs
[    0.265785] usbcore: registered new interface driver hub
[... 276 lines cut ...]
Message from syslogd@raspberrypi at Jun 27 05:35:35 ... 60000113 d7c78564 0000001f c0914e10 00000000
 kernel:[  140.161740] 9f20: d5349f44 d5349f30 c005ec70 c005ebdc 00144311 c016211c d5178480 00c00010
Message from syslogd@raspberrypi at Jun 27 05:35:35 ...
Message from syslogd@raspberrypi at Jun 27 05:35:35 ... d5349df8 c00edfe8 c00ec1d0 00000619 ceebab80
[... cut ...]
 kernel:[  140.161830] 9fe0: 003782fc becf9bc4 000362d4 b6d9f8ac 80000010 00000008 6e616d72 005f5f74ext Page    M-/ Last Line   M-] To Bracket  M-^ Copy Text   M-{ Unindent Text
Message from syslogd@raspberrypi at Jun 27 05:35:35 ...
Message from syslogd@raspberrypi at Jun 27 05:35:35 ... 00000018 00003a98 00c00010 0000007e 00000000
 kernel:[  140.162219] Code: e58c0014 e7842003 e3a03034 e51b2040 (e0212193) 
Message from syslogd@raspberrypi at Jun 27 05:35:35 ...
 kernel:[  140.161690] 9ec0: d5225f60 c004ff58 d5349ec8 d5349ec8 d5349f0c ae214008 d0062028 cee0edc0`

burtbick, 2018-07-06:
Check out my comment to #418 in the forum.  [](https://github.com/google/aiyprojects-raspbian/issues/418) 

I was able to resolve the same type of problem at least for now by explicitly setting the CPU speed in config.txt to less than 1 GHz.

The joy app is working and I'm not getting any lockups of kernel oops failures.

Hopefully this will help while Google tries to figure out if it is a bad batch of pi w boards or the bonnet that doesn't like to run at 1 GHz.

dmitriykovalev, 2018-07-07:
@burtbick Thank you for the detailed comment! VisionBonnet has it's own SoC and clock source, so most likely that's a low level Pi issue (root cause is still unclear). Usually everything works just fine but some boards have issues. By the way, did you try `over_voltage=4` instead of changing CPU frequency?

burtbick, 2018-07-07:
Thanks Dmitriy,

Yes, I tried the over_voltage=4 setting first.  That didn't make ANY difference at all.  I also tried adding over_voltage_min=4 as was suggested in another thread and that also didn't make any difference.

The only thing that seems to help is setting the CPU speed to less than 1 GHz..  Note that the Pi W seems to keep running just fine at 1 GHz with other applications.  It is only when the vision related applications are running that it starts choking.

If I kill the joy demo then I can run the Pi W at full speed and it seems to work with no kernel oops lock ups.  By dropping the Pi speed then the vision demos appear to work just fine, but I haven't run for more than an hour.  With everything back in the closed box and running a video app the temp appears to stabilize out at 62.7 degrees C.

When I get some more time I'll try swapping in a different Pi W and see if that makes any difference.  And I plan on also trying a different SD card to see if that makes any difference.

lufiaraujo, 2018-08-09:
My kit was freezing too, but right after boot if I opened terminal or anything else. Here's what I've done:

1. Shutdown the pi
2. Opened the box and disconnected the camera from the bonnet
3. Booted up
4. Added the "over_voltage" commands to /boot/config.txt:

sudo nano /boot/config.txt

add the following lines to the end of the file:
_over_voltage=4
over_voltage_min=4_

Ctrl+O (save); Ctrl+W (exit)

5. updated the distro

sudo apt-get dist-upgrade
(takes a little forever to download and install everything)

Now it's flawless!

dmitriykovalev, 2018-08-10:
@lufiaraujo Right, that solution works and already described in https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/docs/vision.md.  

@burtbick Can you try to decrease CPU frequency (check #418 for details)?  Looks like that also works sometimes.

seth-johnson-sp, 2018-12-08:
As per the image titled "aiyprojects-2018-11-16.img.xz" I can confirm that the original problem of the OS crashing is solved by adding the following to /boot/config.txt

over_voltage=4
over_voltage_min=4

Do not run the upgrade afterwards. Don't run the upgrade before, either. If you upgrade raspbian you will lose the AIY software functionality with this AIY disk image.