google/aiyprojects-raspbian #348: Two VisionBonnets on RPi

kodukulav, 2018-05-01:
Hi @dmitriykovalev @weiranzhao ,

For my project, I'd like to interface two vision bonnets to RPi. I studied the existing interface and it seems the Pi communicates with Bonnet using SPI protocol. Going a bit deeper, Pi has one dedicated SPI interface and two chip select signals. So, technically, we can connect two SPI devices to the Pi and arbitrate the data transfer between the devices. I'm building a physical system with the aforementioned configuration. 

My question is: Is there an API on the software side to arbitrate between the devices?

Thanks,
Venky

lleonid, 2018-05-01:
There're couple more gpios used in that communication beyond spi lines. And
current kernel driver isn't designed to support multiple bonnets.
For your case it might be easier to just use two pi0 boards with two vision
bonnets (one per pi). And then use networking to communicate between Pi's.

kodukulav, 2018-05-01:
Hi @lleonid 

Thanks for your response.

(1) Is the current kernel driver open sourced? How hard is it to modify the current kernel driver to support multiple Vision Bonnets?

(2) Would it be possible to share the details of GPIO pins that are used for communication between Pi and Vision Bonnet?

(3) For my project, I want to run distributed inference of a neural network between two VisionBonnets. Ideally, I'd like to have a single operating system/host to control the Vision Bonnets. I wonder if this can be possible if I connect two Pis using a ethernet cable.

Thanks,
Venky

lleonid, 2018-05-02:
1. Yes, driver is open source (look for aiy-vision.c under /usr/src).
I would say it's going to be quite time consuming to support multiple
bonnets in driver.

2. It's a bit complicated as one gpio is controlled via i2c command to mcu
on bonned board and the other one is regular rpi gpio. You could look up
details in the driver source.

3. Not sure I understand your concerns here. You can setup network
communication between two Pi's and then there's nothing preventing you from
controlling one pi (and the bonnet attached to it) from another pi. Or you
could control both from another host, being that desktop computer or
something else.

Thanks, Leonid.

dmitriykovalev, 2018-05-02:
Pi Zero has Ethernet-other-USB capability. One option is to connect one Pi Zero into another one using USB cable (and USB OTG adapter). Another option is to connect both of them into network switch/hub.   There are other options, everything probably depends on your use case.

kodukulav, 2018-05-03:
Thanks for the pointers. Another option is stack the VisionBonnets on Pi3 boards and connect the Pi3 boards using a ethernet cable. I'm closing this issue for now. One can always re-open this if they want.

closed 2018-05-03 by kodukulav