google/aiyprojects-raspbian #349: VisionBonnet on RPi3

kodukulav, 2018-05-01:
Hi,

I realized that RPi3 has the same GPIO header as RPi0. So, I'm wondering if I connect VisionBonnet to RPi3, it should work or not.

Thanks,
Venky

dmitriykovalev, 2018-05-02:
It works with RPi3 but you need special flex cable (which is not included in the kit) to connect RPi3 and VisionBonnet.

kodukulav, 2018-05-02:
Got it. Is there any name or standard specification of that flex cable, so that I can buy online?

funwithforks, 2018-05-07:
I bought cables on amazon and the USB was too close and the cable I bought is inflexible at the ends. So I bought an extender for the headers.
https://www.amazon.com/gp/product/B01M6XSBQC/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B0756KM3GB/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
Anyhow, waste money like I did getting a cable that won't fit with the usb in the way or get a cable that is flexible at the ends.

[image]

dmitriykovalev, 2018-05-07:
Actually, your setup should work. Just flip the cable coming from the Pi to VisionBonnet inside VisionBonnet connector (that's a tricky thing and it's different from how Pi Zero is connected). Flex contacts are facing up now, they should be facing down.  VisionBonnet to Camera cable is correct.

hehgoog, 2018-05-09:
What you have should work.
Here is a picture of a setup that worked with RPi3.
[image]

Please note that it does not work with the RPi3B+. To get it working with RPi3B+ you would need to rework a pair of 0201 of resistors and update the mcu firmware.

hehgoog, 2018-05-09:
Not sure if it matters, would need to confirm with schematic, but you should use the cable we shipped between the pi and vision board.

Also I think the cable between your pi and vision kit has one side flipped.  The copper fingers/pads should face into the pcb board on both sides.

closed 2018-06-07 by dmitriykovalev