google/aiyprojects-raspbian #350: ISP on Vision Kit

kodukulav, 2018-05-01:
Hi,

From RPi and Myriad2 documentation, I think there are ISP (image signal processor) blocks on both of them. So, in case of Vision kit, which ISP is being used before feeding the image to the inference engine? Does this vary with CameraInference or ImageInference?

Thanks,
Venky

lleonid, 2018-05-02:
With ImageInference there's no isp being run because you're supplying the
imagen in rgb form.
For camera inference there're two distinct ISPs which are being run in
parallel on myriad and rpi.

Thanks, Leonid.

kodukulav, 2018-05-02:
Hi @lleonid,

Thanks for your response. For CameraInference, why do we need two ISPs to run in parallel, if they are doing the same job? It's waste of computation right.

Thanks,
Venky

dmitriykovalev, 2018-05-02:
ISP on the Pi cannot be used on Myriad chip.  Myriad chip has access only to RAW data (directly from the image sensor), that's why another ISP is required.

kodukulav, 2018-05-03:
@dmitriykovalev : Okay, if that's the case, what is the Pi's ISP used for? In other words, if Myriad2's ISP can convert RAW data to RGB, we don't need another ISP to do the same operation on Pi right.

dmitriykovalev, 2018-05-03:
@kodukulav  We do not modify Pi board and it expects RAW data from the camera sensor. Then it converts RAW data to RGB. Myriad cannot access RGB data on the Pi board, that's why there is another ISP pipeline. You can only see RGB from the Pi on the monitor. RGB from Myriad is not directly visible and only used as an inference input.

kodukulav, 2018-05-04:
Got it. Thanks. I'm closing this issue.

closed 2018-05-04 by kodukulav

virtualjmills, 2018-09-08:
@dmitriykovalev -- Would it be accurate to describe this configuration from a block-diagram perspective as the Myriad "tapping" the RAW datastream as it passes between PiCamera and Pi, so both Myriad ISP and Pi ISP think they're directly fed the RAW datastream from the PiCamera?

Electrically it's obviously a bit more complicated.  :-)

dmitriykovalev, 2018-09-10:
@virtualjmills Yep, you description is pretty accurate.