google/aiyprojects-raspbian #351: ERROR:__main__:AIY VisionBonnet is not detected

jiqin822, 2018-05-02:
I followed the instruction here to assemble the kit:
https://aiyprojects.withgoogle.com/vision#assembly-guide

The privacy light never lits on start up, so I connected a monitor to it and did "sudo systemctl status joy_detection_demo.service":
I got "ERROR:__main__:AIY VisionBonnet is not detected".

I also tried the trouble shooting tips from https://aiyprojects.withgoogle.com/help#booting-and-configuring.

"cat /sys/bus/i2c/devices/1-0051/status_message" I don't have "1-0051" under /sys/bus/i2c/devices/.
"cat /sys/firmware/devicetree/base/hat/product`" I don't have hat under /sys/firmware/devicetree/base/

dmitriykovalev, 2018-05-02:
Looks like VisionBonnet is not detected by Pi for some reason. Can you try to reboot and run again `cat /sys/firmware/devicetree/base/hat/product` ?  Are you using Pi Zero with installed 40-pin header or you soldered it yourself?

jiqin822, 2018-05-14:
Sorry for not getting back to you until now.
I am using Pi zero with installed 40 pin header, the one that comes with the kit.
I rebooted several times, there is no "hat" under /sys/firmware/devicetree/base.
I strictly followed, https://aiyprojects.withgoogle.com/vision#assembly-guide, double checked that everything is connected according to the instruction.

dmitriykovalev, 2018-05-14:
That's pretty weird, can you try to run the following command:

```
sudo dtoverlay aiy-visionbonnet
```
and then copy output from

```
dmesg | grep -i myriad
```

If you see "Myriad ready" message, you can try to restart JoyDemo service and check if it works or not:
```
sudo systemctl restart joy_detection_demo.service
```

dmitriykovalev, 2018-06-07:
Please reopen if you can provide requested details.

closed 2018-06-07 by dmitriykovalev