google/aiyprojects-raspbian #355: google-api-core 0.1.4 has requirement requests<3.0.0dev,>=2.18.0

mpember, 2018-05-03:
Since one of my [SDK-related questions](https://github.com/googlesamples/assistant-sdk-python/issues/132) was closed and it suggested that there was an update, I tried performing an update.

I pulled down the latest code from the git branch and when executing install-deps.sh, the following error appears:
**google-api-core 0.1.4 has requirement requests<3.0.0dev,>=2.18.0, but you'll have requests 2.14.2 which is incompatible.**

ndmax, 2018-07-29:
> The latest code no longer includes install-deps.sh

It appears to be included in the `scripts` director in the latest commit of the aiyprojects branch.

(And I'm receiving the same message...)

dmitriykovalev, 2018-08-16:
There should be no any dependency issues on the latest SD card image.

closed 2018-08-16 by dmitriykovalev