google/aiyprojects-raspbian #356: Mic volume too low with 2018-04-13 image

drigz, 2018-05-04:
[Originally reported by bwims](https://github.com/google/aiyprojects-raspbian/issues/307#issuecomment-385962637):

> [..] the volume is too low with 2018-04-13.  
>
> [TestWavFiles.zip](https://github.com/google/aiyprojects-raspbian/files/1967213/TestWavFiles.zip)
> 
> You should notice that the 2017-05-03 version (OriginalAIY )
> produces a much louder, more audible recording than the 2018-04-13 version (NewAIY).
> 
> Something about the changes in the new version produced to incorporate the V2 boards has reduced the volume of the recording to unacceptably quiet levels, and I can find no way to affect them.
> 
> What I would hope that you mightprovide is a means of adjusting the recording level as per alsamixer, or at least a .ini file with the default setting in an editable form.

and [also by blandiant](https://github.com/google/aiyprojects-raspbian/issues/339#issuecomment-386553201):

> I only have a small problem, when i use the "check audio", i say "1 2 3 testing" the sound volume of the recorded audio is low, even when i speak very close to the microphones
> And when I use the assistant example I have to say "ok google" really close to the microphones to activate the assistant, otherwise it doesn't work

bwims, 2018-05-05:
Many thanks for taking this on board!

LondonSi72, 2018-05-20:
+1 volume of the recorded audio is really low, and as a result I have to speak much more loudly to the voicekit to be heard - especially if cancelling an alarm or timer.  With all the undocumented or wrongly documented changes with each new release, together with the mixing of voice and vision code into the same project this is fast becoming a pain rather than fun.  A real pity as it had potential.

blandiant, 2018-06-04:
hi, 
the problem still has not been fixed?
it is really long, 
I have a project completely stopped because of this (i'm starting to regret to have bought it...).

LondonSi72, 2018-06-11:
Any update on this? Hopefully a fix for this will also fix the problem with TTS responses being too loud and impossible to adjust too. Makes the latest image unusable for me.

manoj7410, 2018-06-22:
@LondonSi72 Can you try below steps:

- Boot the kit.
- Start assistant_grpc_demo.py
- Ask the Assistant, "Increase the volume to 100%"

See if it works.

LondonSi72, 2018-06-26:
@manoj7410 - the problem isn't that the assistant is too quiet, it's that the TTS is too loud.  The ALSAMixer volume control has no effect either, so it's not like I can set the assistant to 100% to match the volume of TTS, then bring them both down via the ALSA control, as that doesn't work. This combined with the fact that the microphone amplification seems to be lower in the latest image - meaning I have to practically shout to get it to respond to alarms / timers etc - makes the latest image pretty unusable for me.

coryd5456, 2018-07-24:
Okay, I found a solution for mine. I noticed that all my custom commands seemed to be at the proper volume, but their assistant voice was the thing that was low. To increase the volume of the assistant I said " Okay Google louder". Saying this 3 times got the volume of the assistant to a desirable level. This didn't change the audio test at all. However, the assistant seems to speak at a desirable audio level. It seems this command must be a new feature in the AIY kit. Which specifically controls the audio level of the assistant, and not the rest of the system.

pholynyk, 2018-08-04:
Had the problem of mic record level too low - verified by making check_audio.py use my_path.wav and transferring it to another machine and looking with audacity. Nearly flat line, so I boosted it by 34db and it was good. Went back to voice kit, did the default.conf mods from #313, then installed paprefs, pasystray, paman, pavucontrol, and pavumeter. Ran pasystray from terminal; click the new speaker icon in tray, select Manages, then Devices tab. Click on Source - input hw_0_0, then click Properties box. Set the Volume to about 216% or 20db, then close properties, and close manager. Finally click the icon again and select quit to get the terminal back clean.
Clearly, the Mic Boost isn't working in the 2018-04-13 image, but I don't know why not (yet).

fjulian79, 2018-08-29:
Ping .. Same problem here!
pholynyk's workaround does not work for me as I use my kit without a monitor, just have a ssh session!

fjulian79, 2018-08-31:
Btw .. I have used aiyprojects-2018-08-03.img.xz ... so maybe the Ticket should be renamed in "Mic volume too low since 2018-04-13 image"

manoj7410, 2018-09-03:
@fjulian79 which version of voice kit do you have ?

DamienFontaine, 2018-09-26:
Same problem but i fixed it with :
```sh
$ pactl set-source-volume 1 500%
```
Where 1 is my alsa_input.hw_0_0. You can find this information with : 
```sh
$ pacmd list-sources
```

cyan198, 2018-10-05:
@DamienFontaine solution increases the recording volume. Just find the right % cause of the background noise level. Thanks a lot!

kutchbhi, 2018-10-16:
@DamienFontaine What are you using to record the audio ? Command `pactl set-source-volume 1 500%` works, but pyaudio library still records in a low volume .  Probably because pyaudio doesn't use pulseaudio and pactl is a pulseaudio command.

DamienFontaine, 2018-10-16:
@kutchbhi I am using Golang + Portaudio(v19). May be the solution is to use the Portaudio library..

patoberli, 2020-06-01:
> Same problem but i fixed it with :
> 
> ```shell
> $ pactl set-source-volume 1 500%
> ```
> 
> Where 1 is my alsa_input.hw_0_0. You can find this information with :
> 
> ```shell
> $ pacmd list-sources
> ```

This also solved the issue with my V1 kit and the 2019 image.