google/aiyprojects-raspbian #361: The Myriad was resetting when using my own model

zhoujustin, 2018-05-08:
Hi!
I'm try to deployed my model in VisionKit!
I'm scratch a new model with 160res_0.5 as the previous issues mentioned.
When I deployed the model in VisionKit, the dmesg out that:
[ 7565.530179] aiy-vision spi0.0: Resetting myriad
[ 7567.715881] aiy-vision spi0.0: Writing myriad firmware
[ 7575.416481] aiy-vision spi0.0: Myriad booting
[ 7575.597658] aiy-vision spi0.0: Myriad ready
[ 8028.335782] aiy-vision spi0.0: Slave not ready after 1000 ms
[ 8028.335805] aiy-vision spi0.0: Fatal error, resetting
and when I debug logging, it's output that:
INFO:root:InferenceEngine transport: _SpiTransport
INFO:root:Loading model "object_detection"...
Inference
INFO:root:Image inference with model "object_detection"...
INFO:root:Unloading model "object_detection"...
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 14] Bad address

The firmware version is 
cat /sys/bus/i2c/devices/1-0051/status_message
OK-V0.5
cat /sys/firmware/devicetree/base/hat/product_ver
0x0002

By the way, the demo of the 'out of box' was worked well.
Thanks!

dmitriykovalev, 2018-05-08:
Could you please put an explicit link to the previous issues, so everything is connected.

weiranzhao, 2018-05-08:
@zhoujustin Is this a mobilenet + ssd model? If so, the only config that's supported right now is 
256x256 depthwise 0.125. And you will need this config
https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/embedded_ssd_mobilenet_v1_coco.config

zhoujustin, 2018-05-09:
@dmitriykovalev About this issue https://github.com/google/aiyprojects-raspbian/issues/314
@weiranzhao Thanks, I will try the 256/0.125 today.

zhoujustin, 2018-05-09:
@weiranzhao Seems like I found the problems of it. The trainning should be "embedded ssd". There's another question, how could I generate the "*_anchors.py" file?
Thanks.

weiranzhao, 2018-05-09:
@zhoujustin were you able to run your model on device using embedded ssd?
as for anchors.py, you can try to have a function similar to this
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/mobilenet_based_classifier.py#L26

It reads from a file with labels. label file just list all the labels with correct order, e.g.,
...
dog,
cat,
person,
grass,
...

zhoujustin, 2018-05-09:
@weiranzhao 
pi@raspberrypi:~ $ python3 AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py --model_path /opt/aiy/models/mobilenet_ssd_256res_0.125_bread.binaryproto --label_path Downloads/label.txt --input_height 256 --input_width 256 --input_layer "Preprocessor/sub" --output_layer "concat,concat_1"
WARNING:root:AIY library supports firmware version 1.1, current firmware version is 1.0. Consider upgrading firmware.
Traceback (most recent call last):
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 137, in <module>
    main()
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 115, in main
    args.threshold, args.top_k)
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 43, in process
    assert len(result.tensors) == 1
AssertionError
pi@raspberrypi:~ $ python3 AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py --model_path /opt/aiy/models/mobilenet_ssd_256res_0.125_bread.binaryproto --label_path Downloads/label.txt --input_height 256 --input_width 256 --input_layer "Preprocessor/sub" --output_layer "concat"
WARNING:root:AIY library supports firmware version 1.1, current firmware version is 1.0. Consider upgrading firmware.
Traceback (most recent call last):
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 137, in <module>
    main()
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 115, in main
    args.threshold, args.top_k)
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 43, in process
    assert len(result.tensors) == 1
AssertionError
pi@raspberrypi:~ $ python3 AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py --model_path /opt/aiy/models/mobilenet_ssd_256res_0.125_bread.binaryproto --label_path Downloads/label.txt --input_height 256 --input_width 256 --input_layer "Preprocessor/sub" --output_layer "concat_1"
WARNING:root:AIY library supports firmware version 1.1, current firmware version is 1.0. Consider upgrading firmware.
Traceback (most recent call last):
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 137, in <module>
    main()
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 115, in main
    args.threshold, args.top_k)
  File "AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 43, in process
    assert len(result.tensors) == 1
AssertionError

zhoujustin, 2018-05-10:
@weiranzhao 
I confused that the binaryproto was created by bonnet_model_compiler. When I deployed it to the vision kit, I got the "len(logit_scores): 10224" assert.
I'm dump both my pb and the person_dog_cat's pb, There are some different between of that. 
There is my model https://drive.google.com/file/d/1bvc6IPwiS-nERJUOErOXyYFr5mHDPgDE/view?usp=sharing
The my.pbtxt is export from frozen pb, and the person_cat_dog.pbtxt is export from person_cat_dog.pb. 
Thanks!

closed 2018-05-11 by zhoujustin