google/aiyprojects-raspbian #362: WiFi problems and VisionBonnet doesn't go into pairing mode

Hofi2010, 2018-05-10:
When I press the button on the VisionBonnet for 5 seconds or longer the led does not come on or start blinking as described in the instructions for connecting the AIY VisionKit to WiFi. The Vision Kit is fully assembled and seems  to be working.

I also cannot connect the AIY Vision kit to a Wifi network manually. Meaning I log into the Raspberry PI with Monitor, keyboard and Mouse attached. In the upper righthand corner where usually the WiFi connections are listed, when you click on the WiFI icon, are two red crosses display when I move my mouse over the icon it says "No interface found". 

So I have no way to connect the Vision Kit to my WiFi network.

dmitriykovalev, 2018-05-10:
Could you please share `dmesg` command output after clicking on Wi-Fi icon?

Hofi2010, 2018-05-10:
„No interface detected“

dmitriykovalev, 2018-05-10:
`dmesg` command prints kernel log, it's pretty long output.

Hofi2010, 2018-05-11:
Ok will come back to you on this traveling at the moment

Hofi2010, 2018-05-20:
I have just up-dated to the new firmware and it is now working correctly

closed 2018-06-21 by dmitriykovalev