google/aiyprojects-raspbian #363: Controlling privacy light, buzzer and arcade button

Bstrum36, 2018-05-10:
I have looked at aiy.vision.leds and have not been successful in even turning on the privacy light.

Below is my naive attempt to turn on the privacy led.

also what is the GPIO  pin  for the led so I can do it directly with low level code.  
Also  GPIO for buzzer and Arcade button

is there a basic example for the arcade button.

print ("----- Start AIY_LED +++++")

from aiy._drivers._hat import get_aiy_device_name
from aiy.toneplayer import TonePlayer
from aiy.vision.inference import CameraInference
from aiy.vision.leds import Leds
from aiy.vision.leds import PrivacyLed
from aiy.vision.models import face_detection

from contextlib import contextmanager
from gpiozero import Button
from picamera import PiCamera
import time
print ("+++++ Imports Completed +++++")

leds = Leds()

leds.privacy_on(brightness = 255)
time.sleep(3)

Thanks

dmitriykovalev, 2018-05-10:
Please check our code examples:  https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/leds_example.py

Bstrum36, 2018-05-10:
The example you provide imports:

from aiy.leds import Leds
from aiy.leds import Pattern
from aiy.leds import PrivacyLed
from aiy.leds import RgbLeds

Which are not found...but when I change this to 
from aiy.vision.leds import Leds
from aiy.vision.leds import Pattern
from aiy.vision.leds import PrivacyLed
from aiy.vision.leds import RgbLeds

The demo works.  

I am now going to see what I can do in my own test.

Thanks