google/aiyprojects-raspbian #364: AIY VISION - The class list for the video_capture.py file is unavailable

davesdere, 2018-05-10:
AIY VISION - The class list for the video_capture.py file is unavailable in the code examples.
It's only written <url>.

dmitriykovalev, 2018-05-10:
This file is available in the latest code revision: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/video_capture.py

davesdere, 2018-05-10:
> aiyprojects/src/examples/vision/video_capture.py

All I see is this. How do I get it? "
A full list of possible categories can be obtained here:
<url> "

D A V I D  C Ô T É

Whatsapp: +1 438 863 0586

*"The best way to predict the future is to create it"* - P. Drucker

dmitriykovalev, 2018-05-10:
All classes supported by image classification can be found here: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/vision/models/image_classification_classes.py

That file (specified by `--classfile` option) contains a subset of these classes, we'll put an example and update the doc in the next release.

closed 2018-05-12 by davesdere

lorenzoromagnoli, 2019-02-06:
Unfortunately, the link is broken. I'd love to have the list of all the classes used in the 'image_classification_camera.py' example

haku15, 2019-02-14:
[This ](https://github.com/google/aiyprojects-raspbian/blob/56d40c7c1c5125fcfe2e69ec2b6f27a15fe35ad9/src/examples/vision/video_capture/video_capture.py#L38) comment line in "video_capture.py" refers to "image_classification_classes.py", which does not exist in current release "v20181116".
In previous release " v20180803" the file was still there. So you can go to the older branch ("Branch"-->"Tag"-->" v20180803") and look for "image_classification_classes.py" or just use this direct [link](https://github.com/google/aiyprojects-raspbian/blob/93a6306438492464c25916b6b2f53d8e4750579b/src/aiy/vision/models/image_classification_classes.py#L16)