google/aiyprojects-raspbian #365: AIY Vision Piezo Buzzer and Arcade Button

Bstrum36, 2018-05-10:
How do you control the piezo  buzzer?
How do you test for the Arcade button status?
What is the GPIO pin  for:
* Privacy light 
*piezo buzzer
 *Arcade button (I assume there is one for each color)??
      I see leds.update(Leds.rgb_on(RED)) 
*Are the LEDs controlled by The KTD driver?? Rater than going to individual GPIO pins?

Thanks

dmitriykovalev, 2018-05-10:
There are many example how to use buzzer: https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/src/examples/vision/buzzer

KDT chip control RGB LED (1st, 2nd, 3rd channels) and privacy LED (4th channel). RGB LED is inside the button. Button example code: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/gpiozero/button_example.py

Bstrum36, 2018-05-10:
Ill take a look and look at the  examples

How about the questions on the Arcade Button?

Bstrum36, 2018-05-10:
I'll look at the KTB Chip contro

But

How about a simple couple of lines of code to turn on or off the buzzer
The examples you listed do not make it clear.

Thanks

Bstrum36, 2018-05-11:
how about a simple code example for telling if the arcade button is pressed?
Thanks

dmitriykovalev, 2018-05-11:
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/gpiozero/button_example.py -- this is simple example for the button. You also need to take a look into [gpiozero](https://gpiozero.readthedocs.io/en/stable/) library docs.

Bstrum36, 2018-05-11:
This looks good I'll give it a try.

dmitriykovalev, 2018-05-11:
Regarding the buzzer, there is an example inside https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/_drivers/_buzzer.py 

```
from aiy._drivers._buzzer import PWMController
    
with PWMController(gpio=22) as controller:
    controller.set_frequency(440.00)
    time.sleep(1)
    controller.set_frequency(0)
```

closed 2018-06-21 by dmitriykovalev

Fernando1954, 2019-02-26:
I broke the arcade button while connecting the led, what is the model number to buy a spare?