google/aiyprojects-raspbian #369: AIY VISION kit 1.1 - image_classification.py

davesdere, 2018-05-12:
When using 
`./image_classification.py --input cam.jpg`
I get
```
WARNING:root:LoadModel: Cannot load model during processing.
Traceback (most recent call last):
  File "./image_classification.py", line 42, in main
    inference.run(image), max_num_objects=5, object_prob_threshold=0.1)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 317, in image_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 117, in transact
    raise SpicommOverflowError(PAYLOAD_SIZE)
aiy._drivers._spicomm.SpicommOverflowError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./image_classification.py", line 48, in <module>
    main()
  File "./image_classification.py", line 44, in main
    print('Result %d: %s (prob=%f)' % (i, label, score))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 108, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 101, in close
    self._engine.unload_model(self._key)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 241, in unload_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Cannot unload model during processing.

```

hkstaub2, 2018-05-12:
I am having the same problem as well.

davesdere, 2018-05-31:
Thanks for the answer. 
I flashed the sd card with the new version, with Etcher and then I got bigger problems...
I also did 
`sudo apt-get update`
and
`sudo apt-get upgrade`
and
`sudo rpi-update`
The pi can't even take a picture.
`raspistill -o cam.jpg`
Just doesn't return anything and doesn't even get stopped by `CTRL + C or CTRL +Z`
I have to close Putty.
`dmesg`
Returns
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.43+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1115 Fri May 25 13:54:20 BST 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c09c04d8, node_mem_map d7c8a000
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=XX:XX:XX:XX:XX:XX vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=b0e94ea2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370296K/393216K available (6419K kernel code, 587K rwdata, 1988K rodata, 440K init, 677K bss, 14728K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc064d0ec   (6421 kB)
                     .init : 0xc08ca000 - 0xc0938000   ( 440 kB)
                     .data : 0xc0938000 - 0xc09cae70   ( 588 kB)
                      .bss : 0xc09d07a4 - 0xc0a79d04   ( 678 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24086 entries in 71 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000073] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000165] bcm2835: system timer (irq = 27)
[    0.000743] Console: colour dummy device 80x30
[    0.001329] console [tty1] enabled
[    0.001382] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060344] pid_max: default: 32768 minimum: 301
[    0.060872] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060928] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062256] Disabling memory control group subsystem
[    0.062476] CPU: Testing write buffer coherency: ok
[    0.063600] Setting up static identity map for 0x8200 - 0x8238
[    0.064884] devtmpfs: initialized
[    0.074061] random: get_random_u32 called from bucket_table_alloc+0x88/0x1c4 with crng_init=0
[    0.074987] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.075420] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.075490] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.076863] pinctrl core: initialized pinctrl subsystem
[    0.078245] NET: Registered protocol family 16
[    0.081229] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.087599] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.087670] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.087817] Serial: AMBA PL011 UART driver
[... 354 lines cut ...]
[  858.082265] [<c016d2c4>] (SyS_ioctl) from [<c000fdc0>] (ret_fast_syscall+0x0/0x28)
[  980.960710] INFO: task kworker/0:1:10 blocked for more than 120 seconds.
[  980.960723]       Not tainted 4.14.43+ #1115
[  980.960727] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  980.960736] kworker/0:1     D    0    10      2 0x00000000
[  980.960787] Workqueue: events dbs_work_handler
[  980.960843] [<c0646518>] (__schedule) from [<c06469e8>] (schedule+0x50/0xa8)
[  980.960865] [<c06469e8>] (schedule) from [<c0646d04>] (schedule_preempt_disabled+0x18/0x1c)
[  980.960889] [<c0646d04>] (schedule_preempt_disabled) from [<c0648274>] (__mutex_lock.constprop.3+0x118/0x358)
[  980.960910] [<c0648274>] (__mutex_lock.constprop.3) from [<c06485cc>] (__mutex_lock_slowpath+0x1c/0x20)
[  980.960928] [<c06485cc>] (__mutex_lock_slowpath) from [<c0648628>] (mutex_lock+0x58/0x5c)
[... cut ...]
[  980.961067] [<c04c388c>] (__cpufreq_driver_target) from [<c04c6464>] (od_dbs_update+0xa8/0x170)
[  980.961085] [<c04c6464>] (od_dbs_update) from [<c04c6f38>] (dbs_work_handler+0x38/0x60)
[  980.961109] [<c04c6f38>] (dbs_work_handler) from [<c00392a4>] (process_one_work+0x11c/0x398)
[  980.961126] [<c00392a4>] (process_one_work) from [<c003955c>] (worker_thread+0x3c/0x544)
[  980.961153] [<c003955c>] (worker_thread) from [<c003f458>] (kthread+0x120/0x15c)
[  980.961181] [<c003f458>] (kthread) from [<c000fe6c>] (ret_from_fork+0x14/0x28)
`
Any idea what I should do next?

davesdere, 2018-05-31:
My bad. After the fresh install, the camera connector latch was half closed.
All working now. Thank you.

davesdere, 2018-05-31:
Closing this

closed 2018-05-31 by davesdere