google/aiyprojects-raspbian #370: install aiyprojects on pi zero for voiceKit v1

kutchbhi, 2018-05-13:
I understand this is not supported. But I have the hardware, and would love to try this anyways.
Any tips please ? 

scripts/install-deps.sh has been removed, so not sure what to try.

drigz, 2018-05-14:
The Pi Zero should work with the SD card image from the AIY website - have you tried it?

kutchbhi, 2018-05-14:
This can be closed. I had bad solders . The microphone and the whole setup works so sweet ! Took me 2 days to set it up, but now the fun time begins :) 

I don't have specific instructions to make this work, for anyone looking. Use "arecord file.wav && aplayfile.wav" to see if the microphone and speakers are working. 
 If not , get the voicehat/microphone/aiy drivers. 
After that if need be, mess around with the SDK (IF REQUIRED)

closed 2018-05-14 by kutchbhi