google/aiyprojects-raspbian #372: TTS: Plan for male voice

NicoLivesey, 2018-05-16:
Hi everyone,

First thank you for the amazing work, this project is really wonderful thanks to you.
I wanted to ask if there was any plan to add other voices such as male voice to the text to speech engine?

For the moment I just changed the pitch parameter in order to get a lower voice but I get as expected an unnatural voice. Do you have any idea on how could I solve this?

Thanks for the help!

dmitriykovalev, 2018-05-17:
We use `pico2wave` offline tts engine to generate voice on the Pi. Unfortunately its configuration is pretty limited, changing pitch parameter is probably the best way for now. We are exploring addition of online tts engine https://cloud.google.com/text-to-speech/ which is much more flexible.

NicoLivesey, 2018-05-22:
Ok thanks for the reply I understand the complexity of the thing and furthermore I find that having an offline tts is really enjoyable for the response time. I have tried google tts but the delay generated by online queries makes conversation less spontaneous.

Anyway thank you, I close this issue since this was just an information question.

closed 2018-05-22 by NicoLivesey