google/aiyprojects-raspbian #374: No module name 'google.assistant.library'

theilgaard, 2018-05-16:
When I try to follow hacking.md for the VoiceKit, in the following step 

`cd ~/voice-recognizer-raspi
source env/bin/activate
python3 src/assistant_library_demo.py`

I get the following error.

`(env) pi@raspberrypi:~/voice-recognizer-raspi $ python3 src/assistant_library_demo.py
Traceback (most recent call last):
  File "src/assistant_library_demo.py", line 33, in <module>
    from google.assistant.library import Assistant
ImportError: No module named 'google.assistant.library'`

theilgaard, 2018-05-22:
Having installed the requirements outside of the virtualenv, I get an Illegal instruction error.

```
pi@raspberrypi:~/voice-recognizer-raspi $ python3 src/assistant_library_demo.py
Illegal instruction
```

dmitriykovalev, 2018-06-29:
@theilgaard You are probably trying to run demo on Pi Zero. There are two ways of using assistant: assistant library and assistant gRPC. Assistant library works only on Pi 3 and doesn't work on Pi Zero (it's not powerful enough) -- https://developers.google.com/assistant/sdk/overview. Assistant gRPC works on both Pi 3 and Pi Zero.