google/aiyprojects-raspbian #378: Script install-deps.sh was removed but instructions were not updated

guidodocter, 2018-05-18:
The instruction file HACKING.md refers to a script called install-deps.sh as its first step. Please remove that line.

janvda, 2018-05-19:
Not only the HACKING.md but also the README.md is referring to install-deps.sh

It would be great to get updated instructions on how I can update/upgrade my aiyprojects installation on my raspberry pi.

So do not only remove the install-deps.sh line, but also add the necessary information about update/upgrading an existing aiyprojects installation.

tobyweston, 2018-05-19:
The instructions (from HACKING.md) seem broken now. On a fresh install of Scratch Lite and skipping the `install-deps.sh` bit, I get the following when testing Audio.

```
$ checkpoints/check_audio.py 
Traceback (most recent call last):
  File "checkpoints/check_audio.py", line 28, in <module>
    import aiy.audio  # noqa
ImportError: No module named 'aiy'
```
Running the following didn't help 😢 

```    
sudo apt-get install python3-pip
pip3 install google-assistant-library==0.1.0
```
No `evn` folder is ever created following the instructions in `HACKING.md` but there is mention of it here and there...

tobyweston, 2018-05-19:
As mentioned in #261 switching to the `voicekit` branch and going through those instructions still works.

BillyTheGit, 2018-06-01:
Im having the same problem, can't update because of the env folder and the install-deps.sh not present.

t1m0thyj, 2018-06-02:
See my response in #393 which basically does the same thing as the workaround mentioned by @tobyweston

manoj7410, 2018-08-10:
@guidodocter Updated HACKING.md has been released and it can be obtained from : https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

closed 2018-08-10 by dmitriykovalev