google/aiyprojects-raspbian #380: Trying to Use Pi Zero Voice Bonnet on Pi 3

PatrickUtz, 2018-05-21:
After playing around with Voice Kit V2, I decided to upgrade to the Pi 3 in order to enable "Ok Google" and use other features. However, I am having trouble using the voice bonnet from the Pi zero kit on the Pi 3.  I have installed a clean version of Raspbian stretch and proceeded in following the instructions on HACKING.md (link below) to set up the Pi. Yet after I install the drivers and run `env/bin/python checkpoints/check_audio.py` , it returns `Failed to find the voiceHat soundcard. Refer yp HACKING.md... etc`. I assume this is due to the fact that the drivers are made for the old voice hat. Yet I am wondering if there is any way I can use the bonnet I have from the Pi 0 on the Pi 3. Any form of help would be greatly appreciated, thanks!

HACKING.md: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

dmitriykovalev, 2018-05-22:
Right, just download the latest [SD card image](https://github.com/google/aiyprojects-raspbian/releases) and run it on Pi 3 with Voice Kit v2. HACKING.md is pretty outdated, we are working on a new one.

PatrickUtz, 2018-05-22:
Great! So I went ahead and used the latest SD card image however I have two issues. First, the volume coming out of the speaker is significantly lower than when it was hooked up to the pi zero (I have to have it set to at least 70 to hear anything). Second, the assistant_demo_library.py runs however the hotword "Ok Google" does not work and when I go to exit the program using ctrl+c nothing happens. Let me know what I should do, thanks!

rollynoel, 2018-05-22:
I'm trying to get AIY Voice 2 running on a RPi3B+, and am running in to problems also ( #381 ).
I think there's some kind of deadlock in the calls to sound support.
The only one I've found works well is assistant_grpc_demo.py, & it only works with a button push.
Ctrl+z, or x, or c usually work for me, then often I hear the first few seconds of the stuck audio.

mindflash-ops, 2018-07-12:
I'm having the exact same issue with a RPi3B+.  I can't seem to get it to work consistently, the microphone is completely broken and the button press only works once usually.  Any subsequent attempt fails.

closed 2018-08-16 by dmitriykovalev