google/aiyprojects-raspbian #382: AIY Vision reports OSError: [Errno 62] Timer expired

empiimp, 2018-05-23:
When I run ./face_detection_camera.py, after a minute I get this dump and then it exits immediately if I try to run it again with the same erro..

`Iteration #3477: num_faces=0
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 62] Timer expired

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./face_detection_camera.py", line 74, in main
    for i, result in enumerate(inference.run()):
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
    yield self._engine.camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 257, in camera_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 147, in transact
    raise SpicommTimeoutError
aiy._drivers._spicomm.SpicommTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 62] Timer expired

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./face_detection_camera.py", line 88, in <module>
    main()
  File "./face_detection_camera.py", line 82, in main
    print('Iteration #%d: num_faces=%d' % (i, len(faces)))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
    self._engine.stop_camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 263, in stop_camera_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 147, in transact
    raise SpicommTimeoutError
aiy._drivers._spicomm.SpicommTimeoutError
`
AIY Vision bonnet V1. 
uname -a
Linux raspberrypi 4.14.34+ #1110 Mon Apr 16 14:51:42 BST 2018 armv6l GNU/Linux

empiimp, 2018-05-25:
`pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 Mon Apr 16 14:51:42 BST 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c09be368, node_mem_map d7c8a000
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:46:77:E3 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=02322b55-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370300K/393216K available (6415K kernel code, 587K rwdata, 1984K rodata, 440K init, 677K bss, 14724K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc064c188   (6417 kB)
                     .init : 0xc08c8000 - 0xc0936000   ( 440 kB)
                     .data : 0xc0936000 - 0xc09c8d10   ( 588 kB)
                      .bss : 0xc09ce65c - 0xc0a77ba4   ( 678 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24081 entries in 71 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000028] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000140] bcm2835: system timer (irq = 27)
[    0.000707] Console: colour dummy device 80x30
[    0.000734] console [tty1] enabled
[    0.000769] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060292] pid_max: default: 32768 minimum: 301
[    0.060783] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060801] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062078] Disabling memory control group subsystem
[    0.062239] CPU: Testing write buffer coherency: ok
[    0.063313] Setting up static identity map for 0x8200 - 0x8238
[    0.064513] devtmpfs: initialized
[    0.074641] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.075019] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.075048] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.076364] pinctrl core: initialized pinctrl subsystem
[    0.077706] NET: Registered protocol family 16
[    0.080768] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.087063] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.087079] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.087197] Serial: AMBA PL011 UART driver
[    0.089933] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.090750] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.129599] bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
[    0.131934] SCSI subsystem initialized
[    0.132163] usbcore: registered new interface driver usbfs
[    0.132271] usbcore: registered new interface driver hub
[    0.132473] usbcore: registered new device driver usb
[    0.140804] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-04-16 18:21
[    0.142857] clocksource: Switched to clocksource timer
[    0.227980] VFS: Disk quotas dquot_6.6.0
[    0.228102] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.228396] FS-Cache: Loaded
[    0.228717] CacheFiles: Loaded
[... 172 lines cut ...]
[  113.755198] aiy-io-i2c 1-0051: Driver loaded
[  113.755544] aiy-vision spi0.0: Initializing
[  113.773629] aiy-vision spi0.0: Failed to bind reset GPIO
[  116.033808] pwm-aiy-io pwm-aiy-io: Driver loaded
[  116.034108] aiy-vision spi0.0: Initializing
[  116.040670] aiy-vision spi0.0: Failed to bind reset GPIO
[  116.231028] gpio-aiy-io gpio-aiy-io: Driver loaded
[  116.231423] aiy-vision spi0.0: Initializing
[... cut ...]
[  118.978770] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  120.193223] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[  123.682095] aiy-vision spi0.0: Myriad booting
[  123.863198] aiy-vision spi0.0: Myriad ready
[  141.152986] fuse init (API version 7.26)
[  890.093959] aiy-vision spi0.0: Incoming crc mismatch: slave c976638d vs computed 56c98728
[ 1566.002983] aiy-vision spi0.0: Transaction timed out
[ 1571.043206] aiy-vision spi0.0: Transaction timed out
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ 
`

dmitriykovalev, 2018-05-26:
Something weird is happening, can you try to generate more debug output in the kernel log by running 
```
modprobe -r aiy-vision && modprobe aiy-vision debug=1
```
and then python example followed by `dmesg`? 

That command line enables debug mode in the driver, switching back to normal is
```
modprobe -r aiy-vision && modprobe aiy-vision
```

empiimp, 2018-05-30:
[dmesg_20180529.log](https://github.com/google/aiyprojects-raspbian/files/2055161/dmesg_20180529.log)

Attached.

empiimp, 2018-06-04:
Hi Dmitriy - Just checking if you had a chance to look through the attached dmesg file from my last post.  After rebooting, it does run the face recognition for a random length of time and once it hits the timeout, any run after that fails immediately. Thank you for the help, Paul...

sainttelant, 2018-06-14:
@empiimp ,hi, guy, i got the totally same errors after i ran ./face_detection_camera.py, so how did you get it done