google/aiyprojects-raspbian #383: 403 error | restricted_client

HuguesT, 2018-05-24:
Hello, 

I'm with the v1 voice kit and the lasted image. 
When I try to start assistant_library_demo.py it bring me  "403 [...] Error: restricted_client" (on the webpage)

Can you help me ? 

Thanks!

piekstra, 2018-05-24:
I have the v2 voice kit and the original image and get a similar issue starting
~/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py
It gives me the URL to enter in my browser and waits for me to "Enter the authorization code: " but visiting the URL just results in a 403.

"Unregistered scope(s) in the request: https://www.googleapis.com/auth/assistant-sdk-prototype"

Also somewhat amusing.. the browser tab title is "Error 403 (OAuth2 Error)!!1"

I would love to know how to get past this. Searches have not yielded any useful results yet and I had wanted to get started using the kit.

HuguesT, 2018-05-24:
i do a fresh install, and everything is ok. Very strange.

piekstra, 2018-05-25:
I went to the site just now and everything worked out.. wonder if there was just a fluke on Google's side. Very strange. I think the issue can be closed as not reproducible.

closed 2018-05-26 by dmitriykovalev