google/aiyprojects-raspbian #384: AIY Project Vision Kit doesn't work

matthewycpang, 2018-05-25:
Hi, 
I have tried the example for my Vision Kit, but it doesn't work. I have followed the guideline from:
https://blog.hackster.io/teething-troubles-for-the-new-aiy-projects-vision-kit-625ed4e9287a

$ cat /proc/cpuinfo | grep Revision
Revision	: 9000c1
$ sudo systemctl stop joy_detection_demo.service
$ sudo rmmod aiy_vision
rmmod: ERROR: Module aiy_vision is not currently loaded
$ sudo rmmod adc
rmmod: ERROR: Module aiy_adc is not currently loaded

Does it mean that there is error in booting the system? I have used the SD Image of 03/01/2018.

Many thanks for your help!

Matthew

dmitriykovalev, 2018-05-26:
Could you please share which example code did you try to run and which error message did you get?

dmitriykovalev, 2018-06-07:
Please reopen if you can provide requested details.

closed 2018-06-07 by dmitriykovalev

novsub, 2019-08-09:
Hello,

I have exactly the same problem, do you have a solution?

Thank's