google/aiyprojects-raspbian #385: AIY Voice Kit V2 and Arcade Button LEDs break after apt-get update/upgrade

TrylN3rr, 2018-05-25:
Similar in nature to this previous issue #223, 
where by after flashing the image from [AIY help site](https://aiyprojects.withgoogle.com/help/#i-want-to-reflash-my-sd-card) to my sd card and then running `~/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py` the arcade button LEDs remains off/do not turn on; however, if I run `~/AIY-projects-python/src/examples/vision/leds_example.py` the arcade button LEDs light up as intended.

If I update the sd card to ensure I have the latest from the AIY project as per the [AIY help site](https://aiyprojects.withgoogle.com/help/#updating-your-sd-card), I get the same errors that plagued #223.  In short, after the kernel update was pushed to stable channel, since I learned from #223 to not run `rpi-update`, the leds-ktd202x driver is gone from `/sys/class/leds` folder, and the `leds_example.py` when run gives the following error 

> FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/ktd202x:led1/device/reset'

Two problems exist:

- I am unsure how one can, or if it's possible to, update the deb packages while skipping an update to the kernel.

- I followed #223 and tried to rebuild the drivers to no avail. I am unsure where to go from here, my thoughts are that the `~/drivers-raspi/leds` may have a role to play, unless the leds-ktd202x driver is deprecated.

TrylN3rr, 2018-05-26:
# Solved
following steps below running `leds_example.py` should light up arcade button on the newly released voice (v2) and vision (v1.2?) kits with rpi zero w. 

[background](##background)  
[the fix](##the_fix)  
[why?](##why?)  

## background
- v2 AIY **voice** kit
- started with latest AIY image (as of now 2018-04-13) from  [AIY help site](https://aiyprojects.withgoogle.com/help/#i-want-to-reflash-my-sd-card)
- `leds_example.py` worked as intended: arcade button lights up and cycles color
(the assistant_grpc_demo.py, mult reports of no arcade button light up, so assumed arcade button leds not supposed to work in that demo)
- via cli ran the **appropriate** (did NOT/do NOT use `rpi-update`) update commands from  [AIY help site](https://aiyprojects.withgoogle.com/help/#i-want-to-reflash-my-sd-card)
- kernel was updated unintentionally & unwillingly, because this was a known issue #223 to break arcade LEDs  
- ran reboot cmd

## the fix
once the raspberry pi finishes reboot cycle 

1. go to `cd /sys/class/leds`, type `ls`, 
if only `led0` dir, then **leds-ktd202 driver** failed to remake and reinstall during kernel update which may be expected [(discussed below](##why?))
2. navigate to `cd /usr/src/`, type `ls`
`leds-ktd202x-1.0` dir should be listed, 
if **not** in dir follow step 3 
if in dir skip step 3, go to step 4
3. to create `leds-ktd202x-1.0` dir: `cp -r ~/drivers-raspi/leds /usr/src/leds-ktd202x-1.0` go into this newly created folder, run `ls`, should see these
> dkms.conf  leds-ktd202x.c  Makefile  make.sh 
4. run `sudo dkms install leds-ktd202x-1.0`
5. when completed run `sudo reboot`
6. when finished rebooting, go to `cd /sys/class/leds`, type `ls`, should see
> ktd202x:led1  ktd202x:led2  ktd202x:led3  ktd202x:led4
7. go to `~/AIY-projects-python/src/examples/vision`, type `leds_example.py`
8. arcade button lights up and cycles through color = success

## why?
- led driver, may have been accidentally left off aiy 1.1 dkms.conf, as I do not see reference to leds-ktd202x there, nor the dir `leds`?  
or
- led driver may be deprecated and there was some other way to activate LEDs, but the examples weren't updated to make use of new LED driver?

dmitriykovalev, 2018-05-26:
@TrylN3rr Thank you for reporting this! We need some time to investigate. KTD driver is *not* deprecated and should still be working after system upgrade.

TrylN3rr, 2018-05-26:
@dmitriykovalev you're welcome. In fact, it was a great learning experience as I'm fairly new to all this,
learned a bit of driver module management (dkms) and debian pkg creation (dpkg) through it all.

markshancock, 2018-05-29:
I have the same symptoms.  I am a new python/pi user.
I was playing around and not I can't get the led to work when I run cloudspeech.py but it does work when I run leds_example.
I have been paying around a bit so I am not sure what broke things,  I ran an update to install VNC (but I do not recall doing an upgrade) and then after that the voice examples were not turning on the led.

@TrylN3rr or @dmitriykovalev can you give me some ideas on how to trace the cause.  My AIY is currently broken.

markshancock, 2018-05-29:
I have played a bit in Python on a couple of other projects and this sort of update/version fragility seems to be inherent to Python.

TrylN3rr, 2018-05-29:
@markshancock in my solved post, under background subheading point three: the assistant_grpc_demo.py, mult reports of no arcade button light up, so assumed arcade button leds not supposed to work in that demo

I believe, refers to the same issue you’re having with cloudspeech_demo.py.

As far as I can tell the demos are referencing the on board led of the voice hat (green led), and not the arcade button. 

now that you mentioned this I’m trying to get arcade button leds  to work for both assistant_grpc_demo and cloudspeech_demo.

markshancock, 2018-05-31:
@TrylN3rr and @a54144 
My perception that cloudspeech_demo.py should control the LED in the Arcade button comes from several YouTube videos that show exactly that behavior.
https://youtu.be/kCfkHkl00gg
https://youtu.be/Y8joFK2O8Es
These seem to be both for V1 rather than the V2 I have.  I wonder if something changed?
I also seem to seeing the light come on before; but, that could be a false memory.
After it was pointed out that leds_example.py worked, I looked at and compared the code and noticed that they were using a different class and method.  It was on my list to investigate.  The whole thing does seem pretty poorly documented.  One frustration is that since V2 uses a Pi Zero, you can't use the "OK Google" alert phrase (one of the original intents of the project).

wejradford, 2018-06-23:
As another data point, I ran a full `apt` update and upgrade and had the same issue: `leds_example.py` worked, but they didn't work when I ran `assistant_grpc_demo.py`.

I looked at @a54144's linked script, but found I that `leds` had moved below `aiy.vision` in the version of the code that I have. 

The following snippet works for me and can be hacked-in in-place of `aiy.voicehat.status_ui`
```python
import aiy.vision.leds
WHITE = (0xFF, 0xFF, 0xFF)
leds = aiy.vision.leds.Leds()
leds.update(leds.rgb_on(WHITE))
```

dmitriykovalev, 2018-11-17:
We released [new SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116) with many fixes in debian package scripts. I hope previous issues should be resolved.

closed 2018-11-17 by dmitriykovalev

ayman, 2019-01-08:
I had to do the same thing for the buzzer `pwm` - following @TrylN3rr instructions above for that module (on the Vision RPI Zero kit).

sethcohn, 2019-04-22:
Confirming: even with most recent SD card image, if you apt-get upgrade, it'll break things due to kernel change.
running this (or some variant of this) will fix it again:
`sudo dpkg-reconfigure aiy-board-info aiy-bt-prov-server aiy-cwc-server aiy-dkms aiy-io-mcu-firmware aiy-models aiy-overlay-vision aiy-overlay-voice aiy-projects-python aiy-python-wheels aiy-usb-gadget aiy-vision-dkms aiy-vision-firmware aiy-voice-services aiy-voicebonnet-soundcard-dkms leds-ktd202x-dkms pwm-soft-dkms
`