google/aiyprojects-raspbian #387: Change the language of Google Assistant on the Raspberry Pi

Albertox0404, 2018-05-29:
Hi, how can i change the language on my Google Assistant of my Raspberry Pi to Spanish?
My english pronunciation is too bad :(
Thank you :)

Smanar, 2018-06-02:
To resume, you have 2 possibilities.
If you use "assistant_grpc_demo.py" use the command aiy.i18n.set_language_code('es-ES')
For other, you need to use google assitant app, in my case I have used an android emulator to do that.

https://stackoverflow.com/questions/48017507/how-do-i-change-the-language-locale-in-the-google-assistant-sdk

hiramvillarreal, 2018-07-12:
Hi, it there any plans to have the "assistant_library_demo.py" & "assistant_library_with_local_commands_demo.py"  working with spanish too?
I been having great time with "assistant_grpc_demo.py" but I would love to have the other working with spanish with local command.

Thanks in advance.

dmitriykovalev, 2018-07-14:
@hiramvillarreal We rely on Google Assistant Library which unfortunately doesn't support Spanish yet: https://developers.google.com/assistant/sdk/reference/library/languages

joseska, 2018-07-16:
@hiramvillarreal yo lo tengo funcionando en español. tanto el "assistant_library_demo.py" como el "assistant_library_with_local_commands_demo.py". Lo que he hecho es instalar el google assistant python sdk (https://developers.google.com/assistant/sdk/guides/library/python/) y a la hora de configurarlo en la aplicacion del movil lo pongo en español.

una vez hecho esto, tanto si lo inicio con "googlesamples-assistant-hotword --project_id my-dev-project --device_model_id my-model" como si lo inicio con "assistant_library_with_local_commands_demo.py" esta en español.....

Si te digo la verdad no se ni muy bien como lo hice, estaba dando palos de ciego y funciono a la primera en español.

hiramvillarreal, 2018-07-17:
@joseskaMuchas gracias, si ya había intentado ese método por seguía sin salirme la opción de español en móvil deja lo intento en esta imagen de SD que ya llevo variops intentos, también yo  a  veces no se qué hago la verdad lástima que el Git de AIY seh a vuelto tan desordenado, lo intento de nuevo por cierto tengo todo los componentes para hacer este proyecto:

https://www.youtube.com/watch?v=rIYdGAjlJ8E&t=2s

closed 2018-08-16 by dmitriykovalev