google/aiyprojects-raspbian #388: arecord: main:788 audio open error: No such file or directory

tfanget, 2018-05-30:
Hello
I'm stuck on step to test the arecord command with this error :  main:788 audio open error: No such file or directory, even after spent a lot of time on Google to find a solution.

arecord -l
carte 1: Microphone [Logitech USB Microphone], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

aplay -l
carte 0: ALSA [bcm2835 ALSA], périphérique 0: bcm2835 ALSA [bcm2835 ALSA]
  Sous-périphériques: 7/7
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
carte 0: ALSA [bcm2835 ALSA], périphérique 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

home/pi/.asoundrc file
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:0,0"
  }
}

What I'm doing wrong......!

Many thanks for your help

Thierry

tfanget, 2018-05-30:
Assound.conf

pcm.!default {

type asym

capture.pcm "mic"

playback.pcm "speaker" }

pcm.mic {

type plug

slave {

pcm "hw:1,0" } }

pcm.speaker {

type plug

slave {

pcm "hw:0,0" } }

dmitriykovalev, 2018-05-30:
Are you using VoiceKit v1 (Pi 3) or VoiceKit v2 (Pi Zero)?

tfanget, 2018-05-31:
No I'm using a RPI2 
But it does'nt work by following this : https://developers.google.com/assistant/sdk/guides/library/python/embed/config-dev-project-and-account

closed 2018-05-31 by tfanget

xcollantes, 2018-07-30:
Where is your .asoundrc file located?

dmitriykovalev, 2018-08-07:
Please to use the latest image from https://github.com/google/aiyprojects-raspbian/releases  It's based on PulseAudio now, `.asoundrc` is not required anymore.