google/aiyprojects-raspbian #390: AIY Vision kit 1.1 Pi WH - Cannot use Picamera() inside a function or a loop

davesdere, 2018-06-01:
Hi there!
I have been fiddling around the Picamera module a little bit and I noticed that when using Picamera() inside a function or a loop I would consistently get this error message.
```
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x13a49b0 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "funSetup_v3.py", line 388, in <module>
    main()
  File "funSetup_v3.py", line 385, in main
    camera = PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
```
I checked the cables. All good. I have the most recent update, upgrade and firmware update available.
```
sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
```
I even run this for good measure.
`sudo apt --fix-broken install`

I tried increasing the memory, by different increment up to 256. It didn't work.
I also rebooted every time I had this error message.
Between each attempt I also did
`pkill python3`

Then...
I tried the 'with' way.
```
def timelapse():
    with picamera.PiCamera() as camera:
```
I also tried to define the camera outside my function.
```
camera = Picamera()
def recordVideo():
    camera.resolution = (1640, 922)  # Full Frame, 16:9 (Camera v2)
```
Then Python told me the variable was accessed before it was created.

I also tried to give the camera global scope.
```
global camera
camera = Picamera()
def recordVideo():
    camera.resolution = (1640, 922)  # Full Frame, 16:9 (Camera v2)
```
Any workaround I missed about this?

dmitriykovalev, 2018-06-02:
It's definitely not related to inside/outside function or loop. Can you check camera using `raspivid` or `raspistill` command line tools? Did you disable JoyDemo service running in background (it uses the camera and camera cannot be shared between processes)?

davesdere, 2018-06-09:
Hi @dmitriykovalev !
I disabled the JoyDemo (First thing I do on a fresh install, those beeps drive me nuts lol). I will take your word on the scope part because since then I have been going about it differently. I did check the `raspivid` and `raspistill` command line tools and got the same error. 
Coming back to the fact that the camera cannot be shared between processes... I know there's the splitter in the doc for streaming and recording at the same time, but what about when the script crashes (e.g. because of a lack of memory) or in the event of a KeyboardInterrupt (without a try/catch). How would you free the camera? The `pkill python3` doesn't seem to work 100% of the time. Is there a way to free the camera? Also, Is there a way to unload the inference models? Can `inference.close()` be used?

dmitriykovalev, 2018-06-11:
Regarding close() functions: camera.close() frees all camera resources and stops camera, inference.close() frees inference resources (unloads model from the bonnet) and stops inferences. That's why you should typically use "with picamera.PiCamera() as camera" or "with CameraInference() as inference" to have these close() functions called automatically.

Can you try our example code (e.g. face_detection_camera.py), does it work on your setup?

davesdere, 2018-06-15:
Hi Dmitry! Thanks for your comment.

I tried the face_detection_camera.py. Here's the result.
```
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ ./face_detection_camera.py
Iteration #0: num_faces=0  
Iteration #1: num_faces=0  
Iteration #2: num_faces=0  
Iteration #3: num_faces=0
Iteration #4: num_faces=0
Iteration #5: num_faces=0
Iteration #6: num_faces=0
Iteration #7: num_faces=0
Iteration #8: num_faces=0
Iteration #9: num_faces=0
Iteration #10: num_faces=0
Iteration #11: num_faces=0
Iteration #12: num_faces=0
Iteration #13: num_faces=0
Iteration #14: num_faces=0
Iteration #15: num_faces=0
Iteration #16: num_faces=0
Iteration #17: num_faces=0
Iteration #18: num_faces=0
Iteration #19: num_faces=0
Iteration #20: num_faces=0
Iteration #21: num_faces=0
Iteration #22: num_faces=0
Iteration #23: num_faces=0
Iteration #24: num_faces=0
Iteration #25: num_faces=0
Iteration #26: num_faces=0
Iteration #27: num_faces=0
Iteration #28: num_faces=0
Iteration #29: num_faces=0
Iteration #30: num_faces=0
Iteration #31: num_faces=0
Iteration #32: num_faces=0
Iteration #33: num_faces=0
Iteration #34: num_faces=0
Iteration #35: num_faces=0
Iteration #36: num_faces=0
Iteration #37: num_faces=0
Iteration #38: num_faces=0
Iteration #39: num_faces=0
Iteration #40: num_faces=0
Iteration #41: num_faces=0
Iteration #42: num_faces=0
Iteration #43: num_faces=0
Iteration #44: num_faces=0
Iteration #45: num_faces=0
Iteration #46: num_faces=0
Iteration #47: num_faces=0
Iteration #48: num_faces=0
Iteration #49: num_faces=0
Iteration #50: num_faces=0
Iteration #51: num_faces=0
Iteration #52: num_faces=0
Iteration #53: num_faces=0
Iteration #54: num_faces=0
Iteration #55: num_faces=0
Iteration #56: num_faces=0
Iteration #57: num_faces=0
Iteration #58: num_faces=0
Iteration #59: num_faces=0
Iteration #60: num_faces=0
Iteration #61: num_faces=0
Iteration #62: num_faces=0
Iteration #63: num_faces=0
Iteration #64: num_faces=0
[... 238 lines cut ...]
It says that the model is already loaded and therefore it doesn't take a picture when there's a face.
I also tried to call the `inference.close()` from within with `CameraInference(face_detection.model()) as inference:` without any luck. I
```
def main():
    with PiCamera() as camera:
        # Configure camera
        camera.resolution = (1640, 922)# Full Frame, 16:9 (Camera v2)
        camera.start_preview()
        # Do inference on VisionBonnet
        with CameraInference(face_detection.model()) as inference:
            for result in inference.run():
                if len(face_detection.get_faces(result)) >= 1:
                    filename = dt.datetime.now().strftime("%Y-%m-%d_%H.%M.%S.jpg")
                    camera.capture(filename)
                    camera.resolution = (640, 480)
                    camera.start_recording(dt.datetime.now().strftime("%Y-%m-%d_%H.%M.%S.h264"))
                    camera.wait_recording(600)
                    camera.stop_recording()
                    send(filename) # Send the JPG by email.
                    break
            inference.close() #Not working
        camera.stop_preview()
        camera.close()
```
This script also when called from a bash script or as a function called within a Python While loop, it doesn't free the mmal ressources. I will read the doc about that 'with' statement until your next reply.

dmitriykovalev, 2018-06-18:
You don't need to call `inference.close()` explicitly because of `with CameraInference(...)` construction. Otherwise it's going to call `inference.close()` twice and the second one would fail.  You also don't need to call `camera.close()` explicitly for the same reason. 

`CameraInference` is a helper class which loads and unloads model: https://github.com/google/aiyprojects-raspbian/blob/e48363018e37b1fdc22d5ddc4b221db97c2cf47b/src/aiy/vision/inference.py#L68

Having `WARNING:root:LoadModel: Model 'FaceDetection' is already loaded.` is fine, that's just a warning.

davesdere, 2018-06-18:
Thank you Dmitri! I just looked over this bit of code differently because of that conversation.

dmitriykovalev, 2018-06-21:
@davesdere No problem! Closing this for now.

closed 2018-06-21 by dmitriykovalev