google/aiyprojects-raspbian #391: Voice kit v1 with Pi Zero

mpember, 2018-06-01:
With the new kit being released with a Pi0, is this change in compatibility due to a hardware change in the HAT?

Does the current code/API support using the original HAT with a Pi Zero W?

dmitriykovalev, 2018-06-02:
Yes, original (v1) Voice Hat  should work on Pi Zero as well, just use our SD card image.

closed 2018-06-02 by dmitriykovalev

mpember, 2018-06-03:
Does this mean that the limitation of hotword support was resolved?  Are there no functional limitations that result from using a Pi0?

mpember, 2018-06-04:
@dmitriykovalev I have installed the official image.  Only the button trigger works. No sign of hotword trigger working.  Is there any special step required to get it working?

dmitriykovalev, 2018-06-05:
That's correct. Unfortunately, Pi Zero is not powerful enough for real time hotword detection. We rely on assistant library which is optimized only for armv7l, i.e. for Pi 3, more info can found here: https://developers.google.com/assistant/sdk/overview

mpember, 2018-06-05:
@dmitriykovalev 
Does this mean that v2 of the kit is less functional than v1?  Does the v2 image use a different API/SDK?

dmitriykovalev, 2018-06-05:
No, that doesn't mean that. V2 board is more advanced than V1. The question is about Raspberry Pi board (Pi Zero vs Pi 3). We use the same software and the same Assistant library for both.

mpember, 2018-06-05:
So, anyone who owns a v2 voicekit would need to replace the Pi0 (and the case) to get hotword functionality?  Am I wrong to consider the loss of hotword functionality a step backwards?

PeterMalkin, 2018-06-05:
@mpember 
Not exactly.
Voice Hat v2 and Voice Hat v1 both work on either version of the raspberry pi. So users feel free to use either of the Hats with the raspberry pi of their choosing.
Raspberry pi 0 is not powerful enough to support hotwording. This is due to the fact that their CPU is only single core arm v6. AIY team cannot really do anything about it, therefore, if you want to use the hotwording feature you have to use raspberry pi 3 with your builds.

Hope this helps.