google/aiyprojects-raspbian #394: "Unexpected error" when running bonnet_model_compiler.par on "mobilenet_v1_160res_0.5_imagenet.pb"

mircea800, 2018-06-02:
As a fist step, I've been trying to run the latest bonnet_model_compiler.par using "mobilenet_v1_160res_0.5_imagenet.pb" (both located in the same folder on my machine and downloaded using the links on aiyprojects.withgoogle.com/vision/).  

I run the compiler using:

./bonnet_model_compiler.par  --frozen_graph_path=./mobilenet_v1_160res_0.5_imagenet.pb --output_graph_path=./mobilenet_v1_160res_0.5_imagenet.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Softmax"  --input_tensor_size=160 

but it always returns an "unexpected error".  I tried using the --debug flag, but I can't find the crash log.  Setting the flag does create a folder called "bonnet_model_compiler.runfiles-mircea" in the /tmp directory, but it I don't see any log files in there.

I've tried the above running 64-bit versions of Ubuntu 14.04.5 LTS, Ubuntu 16.04 and Lubuntu 14.04 installed on a 2007 Macbook Pro, but with no luck.  In each case, Ubuntu/Lubuntu was the only the OS installed (not a virtual box) and I made sure that Python 2.7 was the default version being used.  

Can the Macbook Pro hardware be the problem or is there something else I'm missing in the setup?  I checked issue #314, but did not see an answer that would help troubleshoot my problem any further.   

Thanks,
Mircea

weiranzhao, 2018-06-07:
I've never run into such issue before. Could you try with python3?

mircea800, 2018-06-10:
Just tried it with Python 3.4.3 and still no luck.  Also tried another fresh install of Lubuntu 14.04 followed by installing Python3, pip3 and Virtualenv and same result.  Any particular packages that need to be installed?

dmitriykovalev, 2018-08-16:
@mircea800  Are you still having this problem?

mircea800, 2018-09-14:
I just downloaded another copy of bonnet_model_compiler.par and the latest mobilenet_v1_160res_0.5_imagenet.pb, but same problem with an "unexpected error".  And still no log file in the "bonnet_model_compiler.runfiles-mircea" subfolder in the /tmp directory.

weiranzhao, 2018-09-14:
@mircea800 

I tried again on Ubuntu 16.04, installed on a VirtualBox. It works fine. It's hard to imagine this is caused by running Ubuntu on MBP directly. But would you mind to try again with Ubuntu installed in VirtualBox?