google/aiyprojects-raspbian #395: Need to stream the face detected video over network

titusece, 2018-06-02:
Hello,
I would like to stream the face detected camera output to another raspberry pi board over network.

Typically I will use the below commands to stream the raspberry pi camera output to another raspberry pi board.

> raspivid -t 999999 -h 720 -w 1080 -fps 25 -b 2000000 -o - | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! gdppay ! udpsink host=192.168.0.89 port=5000
> 
> gst-launch-1.0 -v udpsrc port=5000 ! gdpdepay ! rtph264depay ! avdec_h264  ! videoconvert ! fbdevsink sync=false

Want to stream like this but I need face detected processed output by google vision kit
Any help on this ?

dmitriykovalev, 2018-06-02:
You have to program the solution :)  Use our python API to get inference data from face detection model and then then somehow attach to video stream generated by the camera. There are many possible ways to do that, probably depends on your use case and software you want to use on the receiving end.

dmitriykovalev, 2018-06-07:
Please reopen if you have other questions.

closed 2018-06-07 by dmitriykovalev