google/aiyprojects-raspbian #397: Get wrong color from picamera API

rouyunpan, 2018-06-04:
I modify the AIY example "face_detection_camera.py" to do Face recognition. 
https://youtu.be/zFWgmFdt5Dg

I get the raw data of a frame with the picamera API camera.capture() and do Face recognition with OpenCV.
The execution time is a little slow and sometime I  got the wrong color.
do you have any suggestion for this situation and performance?

[My code]
with PiCamera() as camera:
 # Start the camera stream.
       camera.resolution = (640, 480)
        camera.framerate = 30
       
        rawCapture = PiRGBArray(camera)
        time.sleep(1)
        frame_count = 0
        
        image = np.zeros((480, 640, 3), np.uint8)
        
        with CameraInference(face_detection.model()) as inference:
            for i, result in enumerate(inference.run()):
                faces = face_detection.get_faces(result)             
                camera.capture(rawCapture, format="bgr", use_video_port = False)
                image = rawCapture.array.copy()
        ....
        cv2.imshow('Face recognition', image)

dmitriykovalev, 2018-06-05:
Getting wrong color is pretty strange, I've never seen something like this with Pi camera. I think you should double check your image processing pipeline. Can you reliably reproduce this issue without OpenCV?

rouyunpan, 2018-06-08:
if I only call picamera camera.capture() without calling CameraInference(). 
The color is normal during the period. 

Any time sync that I must care about for  CameraInference()?
Or you have any example that we can get the raw data for a frame during CameraInference()?

---------------------------------------------------------------------------
camera = PiCamera()
camera.resolution = (640, 480)
camera.framerate = 30
rawCapture = PiRGBArray(camera, size=(640, 480))
time.sleep(0.1)

# capture frames from the camera
while (1):
    camera.capture(rawCapture, format="bgr", use_video_port = False)
    image = frame.array
    cv2.imshow("Frame", image)
    key = cv2.waitKey(1) & 0xFF
    rawCapture.truncate(0)
    if key == ord("q"):
        break

rouyunpan, 2018-06-15:
Hi 

I mortify my code as below, and fix this problem now.
This's our demo for facial recognition with google AIY and inte NCS.
https://youtu.be/zQHOLmRjpek

[My code]
...
    frameSize = (640,480)
    vs = VideoStream(src=0,usePiCamera=True, resolution=frameSize, framerate=32).start() 
    defWidth = 640
    defHeight = 480
    time.sleep(2) 
...     
    with CameraInference(face_detection.model()) as inference:
            for i, result in enumerate(inference.run()):
                faces = face_detection.get_faces(result)
                frame = vs.read()             
                for face in faces:
                    fX, fY, fWidth, fHeight = transform(face.bounding_box)
                    if(fX > 0) and (fY > 0) and (fWidth -fX > 150) and (fHeight -fY > 150):
                        cv2.rectangle(frame, (fX, fY), (fWidth, fHeight), (0,255,0), 1)
                        roiImage = frame[fY:fHeight, fX:fWidth]              
                        frame_count += 1

                        test_output = run_inference(roiImage, graph)

                        if (face_match(valid_output, test_output)):
                            found_match = True
                            cv2.putText(frame, "Rouyun",(fX+5,fHeight+5),cv2.FONT_HERSHEY_PLAIN,1.5,(255,255,255),2)
                        else:
                            found_match = False
                            cv2.putText(frame, "Unknown",(fX+5,fHeight+5),cv2.FONT_HERSHEY_PLAIN,1.5,(255,255,255),2)

                cv2.imshow('Face recognition', frame)                
                key = cv2.waitKey(1) & 0xFF
                # if the `q` key was pressed, break from the loop            
                if key == ord("q") or key == ord("Q") :
                    break
    
    vs.stop()
...

closed 2018-06-15 by rouyunpan

dmitriykovalev, 2018-06-18:
@rouyunpan Very nice project!