google/aiyprojects-raspbian #400: AIY Voice Kit Version 2.0, google assistant examples do not work

firechen, 2018-06-05:
Recently, I have purchased AIY Voice Kit version 2.0 and tried it with my Raspberry Pi 3B. The assistant_grpc_demo.py and cloudspeech_demo.py work well for me. But for those about assistant_library, the demo does not work, there is no response from the demo. For example, I run assistant_library_with_local_commands_demo.py and say "Ok, Google IP address", the program will be blocked there. I have used top command to see the CPU usage, the CPU usage will be above 100. This problem also occurs in other assistant_library demos.

rollynoel, 2018-06-05:
I've fought the same issue to a standstill ( see #381 for what I've determined ). I've communicated with Google AIY support ( support-aiyprojects@google.com ), and await their solution.
I changed to a 3B+ so I could work the AIY without having to cross the room to push its button.
I'm glad it's not only my AIY experiencing this problem.

rollynoel, 2018-06-05:
Further, whilst running these tests, something is consuming the /dev/root space on my micro sd card,
  such that I end up using 100% of it, and the card became unuseable. This was particularly useless with the 8 GB card provided.

Jianan-Li, 2018-06-11:
None of the assistant_library demos work for me either. I'm using a Raspberry Pi 3B with aiyprojects-2018-04-13.img.

firechen, 2018-06-11:
Event though those assistant_library demos do not work, I am able to use custom hotword to trigger grpc assistant now. We can do this with the help of snowboy.

rollynoel, 2018-06-11:
Thanks ever for the [Snowboy lead](http://docs.kitt.ai/snowboy/). Using Snowboy, will it run well on the included Pi-Zero?

firechen, 2018-06-12:
Yes, it works well in both Pi-3b and Pi-Zero WH.

Jianan-Li, 2018-06-12:
Thanks for the snowboy lead. But to be honest, I'm still kind of shocked by the fact that the AIY team at Google released an image with broken examples and haven't released an update to fix it in 2 months, and now we have to use a hacky workaround. Is it just me? Do you guys all think that this is acceptable? Especially considering that we all paid $50 dollars for this kit.

rollynoel, 2018-06-18:
Any special incantations required to get snowboy to work with Python 3?

charles-d-burton, 2018-06-22:
I'm also having issue with this.  When I run the commands on a 3B+ the button starts to blink and it seems like everything is working.  The problem is that nothing happens, if I say OK Google I get nothing and if I push the button and speak to it nothing happens there either.  It seems like the microphone is not working.

Smanar, 2018-06-22:
The step 64 "Check your audio" have worked ?

charles-d-burton, 2018-06-22:
Yeah, output works fine and the repeating the audio back works fine too.  It's just the demo applications that fail.

rollynoel, 2018-06-22:
This is what I last heard June 14th when I reported this issue to Google (initially May 15th):
"We were able to reproduce this issue and have reported it to the developers with the highest priority. We shall get back to you as soon as we get a resolution for this one.

Thanks
AIY Support Team"

Smanar, 2018-06-22:
Lol, I have some prbs too, but with you, I m feeling lucky, I have a Google assistant working on a raspberry 1 B+.
It's realy strange no-one in this communauty have find a solution.

rollynoel, 2018-06-24:
So a AIY Voice 2 hat worked on a RPi with only 26 GPIO pins?

Smanar, 2018-06-25:
The AIY lib yes, I m not using the hardware.
I m using it with a usb sound adaptator https://www.gear4music.com/G4M/Electrovision-USB-Sound-Adaptor/1SD6 with a portable HP + microphone and 2/3 electronics components. Kit is totaly useless.

rollynoel, 2018-07-22:
I just closed & commented on issue #381 with what I believe is a fix from Google AIY Support to eliminate this problem. I hope it works as well for you as it did for me.

KyleMitchel, 2018-07-28:
I also have this issue. I used @rollynoel suggestion on #381 to no avail.

watou, 2018-08-01:
Voice Kit V1 on 3B+, same symptoms as here, [this fix](https://github.com/google/aiyprojects-raspbian/issues/381#issuecomment-406903158) also does not solve the problem of no assistant output.  "Hey, here's a kit that doesn't work, enjoy!" ☹️

watou, 2018-08-01:
@manoj7410 the suggestion on #381 does not work for my RPi 3B+ / Voice Kit V1 / aiyprojects-2018-04-13.img.xz (plus updates) setup.

rollynoel, 2018-08-02:
Sorry - it certainly did the trick for my V2 Voice Kit.

watou, 2018-08-02:
@rollynoel That is at least helpful that it works for V2 and not V1.  I hope a Google person can rescue us V1 Voice Kit users!

manoj7410, 2018-08-10:
@watou A new build has been released by Google and can be obtained from : https://github.com/google/aiyprojects-raspbian/releases Can you please check if your issue got resolved.

rollynoel, 2018-08-10:
It has indeed, thanks!

watou, 2018-08-10:
@manoj7410 With the Voice Kit V1, the new disk image is now dumping huge piles of JSON and HTML into the terminal window, but still not speaking back to me.  I am not performing a `sudo apt-get update ; sudo apt-get upgrade` to see if that makes any difference.  I will now try to decipher what all that text is trying to tell me and report further.

jlawrence124, 2018-08-10:
Same thing on my end @watou

dmitriykovalev, 2018-08-10:
Seems to be this issue in the Assistant library: https://github.com/googlesamples/assistant-sdk-python/issues/274

closed 2018-11-16 by dmitriykovalev