google/aiyprojects-raspbian #403: AIY-Voice autostart when system boots doesn't work

KevinZhou92, 2018-06-10:
Follow the instruction on the official website[Link](https://aiyprojects.withgoogle.com/voice-v1/#users-guide-start-the-assistant-library-demo-app)

However, i kept getting error message like
```
voice-recognizer.service - voice recognizer
   Loaded: loaded (/etc/systemd/system/voice-recognizer.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-06-10 01:19:45 UTC; 12s ago
 Main PID: 2206 (python3)
   CGroup: /system.slice/voice-recognizer.service
           ├─2206 python3 /home/pi/main.py
           └─2215 /usr/bin/pulseaudio --start --log-target=syslog

Jun 10 01:19:45 raspberrypi systemd[1]: Started voice recognizer.
Jun 10 01:19:48 raspberrypi pulseaudio[2215]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jun 10 01:19:48 raspberrypi pulseaudio[2215]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

> The image i use was `aiyprojects-2018-04-13.img`

Does anyone has a similar issue? This is so frustrating.

dmitriykovalev, 2018-06-11:
Please check #295 thread and search for DISPLAY there.

KevinZhou92, 2018-06-11:
@dmitriykovale Thanks for timely response.

I got the solutione from #295. If anyone visited this thread encountered the same error message. The solution is to add a magic line in the systemctl script like
```
[Service]
Environment=DISPLAY=:0
```

This solution is from @divx118. Thanks guys.

dmitriykovalev, 2018-06-11:
@KevinZhou92 perfect!

closed 2018-06-11 by dmitriykovalev