google/aiyprojects-raspbian #406: Latest image (2018-04-13) does not work with Raspberry Pi 3 B+ for Voice Kit V1

ahomeuser, 2018-06-13:
Hi, this is related to #310 where people haven't been able to make pi 3B+ work with Voice Kit V1. I tried using the latest image (2018-04-13), but it doesn't go beyond 4 raspberries and cursor. It seems others were able to get it working, hence the issue was closed. But I found someone else still having issue recently like me. Please help.

dmitriykovalev, 2018-06-13:
Just to confirm, you Pi 3B+ is working fine with  Raspbian image and doesn't work with AIY 2018-04-13 image?

ahomeuser, 2018-06-13:
Thanks for quick response. I was able to figure out the issue. It was not the image, but power supply. I found [this](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) blog which mentioned the lightning symbol. I am now able to get the basic setup working with different power cable.

closed 2018-06-13 by ahomeuser

babyrobotx, 2018-07-26:
Just bought P3B+, Voice Kit V1, installed latest image.
Encountered issue as mentioned in the title.

ahomeuser is correct.

I've tried 3 cables from different vendors:
With Cable1, power pad on P3B+ board read ~5.80V, and system will not boot;
With Cable2, power pad on P3B+ board read ~5.90V, boot and a "lightening" symbol shows up on top right corner above Desktop;
With Cable3, power pad on P3B+ board read ~4.999V, boot and aiyproject works smoothly.

This is just to confirm ahomeuser's conclusion that good quality power cable does matter to P3B+.
Guess that's due to Brown Out function of P3B+ board.

dmitriykovalev, 2018-07-28:
@babyrobotx Thank you for useful details, good power cable is really important.

ilatypov, 2019-05-07:
How to tell a "good quality power cable" on the market?

babyrobotx, 2019-05-07:
That's a good question.
Most convenient way is to use a dedicated device to test impedance, which is on market.
An econimic way is to multimeter to test yourself.
If you think those bother too much, then you should buy a branded cable. They cost higher than unbranded ones, but are still as cheap as below 10 dollars. They are worth the money, since the branded ones use better material, low resistance, higher current carrying capability.
Of the 3 I tried, the only cable that reads 4.999v is a cable from a brand which sells quality products. Another thing to mention, try short cables, for better IR drop.