google/aiyprojects-raspbian #410: Accessing Myriad Development Kit in Vision Bonnet

mikeliao97, 2018-06-14:
Hi, very cool project! Thanks for the hardwork.

So I am aware that the Vision Bonnet is a Movidius 2450, and I am aware that Movidius has a Myriad Development Kit. Do you know if its possible to use the SDK? or follow tutorials using the Intel Compute Stick(which also has a Movidius 2450)?

A beginner question, thanks!

https://uploads.movidius.com/1463156689-2016-04-29_VPU_ProductBrief.pdf

dmitriykovalev, 2018-06-15:
Hi Mike! We do not expose any Myriad internals because of huge complexity. We only have Python API to run compute graphs on Myriad. Intel Compute Stick is a different product, so there is no way to use our example code there.

liyoung1992, 2018-06-15:
@dmitriykovalev Hi，I read your  article today [aiy-vision-kit](https://cogint.ai/custom-vision-training-on-the-aiy-vision-kit/),and I am very admire you ！now，I begin to compile own model，and  I met many difficulties！Do you have any practice in compile ownmodel！For example ：[issues/411](https://github.com/google/aiyprojects-raspbian/issues/411)

closed 2018-06-15 by mikeliao97

elyas20, 2018-06-22:
Hi @dmitriykovalev,
I need to use myriad for traditional image processing and computer vision routines. Is there any dev board that contain the myriad that I can purchase ?
Thanks.

dmitriykovalev, 2018-06-23:
@elyas20 You should contact https://www.movidius.com/ regarding Myriad dev boards.

phaseOne, 2018-08-01:
Hey @dmitriykovalev, I'd love to hack around with the Movidius SDK. I can arrange an NDA if needed.

dmitriykovalev, 2018-08-01:
@phaseOne You still need to contact https://www.movidius.com/ for that. You can also send detailed email to our support team support-aiyprojects@google.com.