google/aiyprojects-raspbian #412: aiyprojects-2018-04-13 not booting on RPI2 (Voice kit V. 1)

luizbalestreri, 2018-06-15:
Hello, I just bought the Version 1 kit and tryed installing the kit with the image provided by AIY.

It successfully started (rainbow screen), it shows the Raspbian boot welcome, some drivers loading, and then when the desktop environment should appears it stays in the terminal screen whit the white "_" blinking (like it's loading something). But it seems to be in some kind of boot loop, because it blink constantly (showing that it's loading), then it disappear for 3 seconds and starts blinking again (like it's restarts the loading). Tryed burning the image several times and the problem persists. Thank you

dmitriykovalev, 2018-06-18:
Can you please try to reflash SD card using our latest image from https://github.com/google/aiyprojects-raspbian/releases? Make sure to use wall power adapter, i.e. at least 2.1 Amps.

luizbalestreri, 2018-06-18:
Hello Dmitty, thanks for your answer. It seems I'm using the most recent version(2018-04-13 ). I'm also using a 2.1 amps charger from my cellphone. I'll try flashing and older one. Thank you!

dmitriykovalev, 2018-06-19:
Sure, let me know how that goes. Reboot loop seems to be very unlikely.

closed 2018-07-28 by dmitriykovalev