google/aiyprojects-raspbian #413: Connection an OLED screen through the HAT pins

smoscar, 2018-06-16:
Hey, this is a great project!

I am not sure if I fully understand this but looking through the code it seems the bonnet hat is using I2C as a communication protocol between the Movidius chip and the extra GPIO pins (possibly to be able to connect multiple devices through one bus).

Is it possible to use this bus to connect an I2C OLED screen to the kit? [Something like this](https://www.ebay.com/itm/0-96-I2C-IIC-Serial-128X64-128-64-Blue-OLED-LCD-LED-Display-Module-for-Arduino/401395129024?epid=19015370270&hash=item5d75039ac0:g:H2QAAOSwR6RZ6d2n) would be perfect in size to display the status of the AIY Kit (turned on, wifi signal, bluetooth, etc.) without being connected to an HDMI monitor.

Thanks for the hard work put into this!

dmitriykovalev, 2018-06-17:
Hi @smoscar, thank you! Our extra GPIO pins are controlled by a separate microcontroller, not the Movidius chip. Raspberry Pi talks to the microcontroller via I2C bus. You can definitely connect more devices to this I2C bus, pinout details here: https://pinout.xyz/pinout/i2c You probably need to use soldering iron to attach to I2C pins on the bonnet 40-pin connector.

smoscar, 2018-06-18:
Thanks for the quick response @dmitriykovalev!

I ended up soldering it to the Pi instead, I figured it would be easier to replace if anything would have gone wrong, but it's working great.

dmitriykovalev, 2018-06-18:
@smoscar Great!

closed 2018-06-18 by dmitriykovalev