google/aiyprojects-raspbian #414: Vision Bonnet Pinout

wuhland, 2018-06-19:
I'd like to use some of RPi's pins to integrate other devices with the AIY Vision project. For instance, I would like to free up the TX, RX, 3.3V, and GND pins to connect with adafruits fona breakout board. Is this possible? Is there a schematic somewhere which lists which pins the vision bonnet uses?

dmitriykovalev, 2018-06-25:
@wuhland sorry for the delay, I'll update soon with the pins used by VisionBonnet.

dmitriykovalev, 2018-06-29:
These are the [40-pin connector](https://pinout.xyz/) pins used by VisionBonnet:

```
                   1    2
       I2C_SDA --> 3    4
       I2C_SCL --> 5    6
                   7    8
                   9   10
     MRD_RESET --> 11  12
                   13  14
   BUZZER_GPIO --> 15  16 <-- BUTTON_GPIO
                   17  18
      SPI_MOSI --> 19  20
      SPI_MISO --> 21  22
      SPI_SCLK --> 23  24 <-- SPI_CE
                   25  26
        ID_SDA --> 27  28 <-- ID_SCL
                   29  30
 PI_TO_MRD_IRQ --> 31  32
 MRD_TO_PI_IRQ --> 33  34
                   35  36
                   37  38
                   39  40
```

closed 2018-07-02 by dmitriykovalev