google/aiyprojects-raspbian #416: SpicommOverflow(PAYLOAD_SIZE) error on custom model

jcorbett, 2018-06-24:
Looking at the _drivers/_spicomm.py, it looks like the image is either too big or too small. Before I retrain the model, can someone verify that 144,144,3 will work? (pathfinding.binaryproto is in /opt/aiy/models)

I get this error just by initiating the model:
```
$ ./pathfinder.py --input test.png
> Traceback (most recent call last):
  File "./pathfinder.py", line 21, in <module>
    main()
  File "./pathfinder.py", line 14, in main
    with ImageInference(pathfinder_model.model()) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 95, in __init__
    self._key = self._engine.load_model(descriptor)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 225, in load_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 117, in transact
    raise SpicommOverflowError(PAYLOAD_SIZE)
aiy._drivers._spicomm.SpicommOverflowError
```
```
$ file ./test.png
> test.png: PNG image data, 144 x 144, 8-bit/color RGB, non-interlaced
```

**pathfinder_model.py**
```
from aiy.vision.inference import ModelDescriptor
from aiy.vision.models import utils

_COMPUTE_GRAPH_NAME = 'pathfinding.binaryproto'

def model():
    return ModelDescriptor(
        name='pathfinding',
        input_shape=(1, 144, 144, 3),
        input_normalizer=(128.0, 128.0),
        compute_graph=utils.load_compute_graph(_COMPUTE_GRAPH_NAME))
```

**pathfinder.py**
```
#!/usr/bin/env python3
import argparse
from PIL import Image

from aiy.vision.inference import ImageInference
import pathfinder_model

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument('--input', '-i', dest='input', required=True)
    args = parser.parse_args()

    with ImageInference(pathfinder_model.model()) as inference:
        image = Image.open(args.input)
        print(inference.run(image))

if __name__ == '__main__':
    main()
```

dmitriykovalev, 2018-06-25:
How big is the `pathfinding.binaryproto`  file? Can you please also attach `dmesg`  output when everything crashes?

jcorbett, 2018-06-26:
```
$ ls -alh /opt/aiy/models/pathfinding.binaryproto
-rwxr-xr-x 1 root root 18M Jun 24 04:57 /opt/aiy/models/pathfinding.binaryproto
```
```
$ dmesg | grep -i aiy-vision
[   20.479438] aiy-vision spi0.0: Initializing
[   20.514260] aiy-vision spi0.0: Failed to bind reset GPIO
[   22.830056] aiy-vision spi0.0: Initializing
[   22.839496] aiy-vision spi0.0: Failed to bind reset GPIO
[   22.990314] aiy-vision spi0.0: Initializing
[   22.995914] aiy-vision spi0.0: Failed to bind reset GPIO
[   26.225906] aiy-vision spi0.0: Initializing
[   26.227433] aiy-vision spi0.0: Failed to bind reset GPIO
[   27.084484] aiy-vision spi0.0: Initializing
[   27.105298] aiy-vision spi0.0: Resetting myriad on probe
[   27.105315] aiy-vision spi0.0: Resetting myriad
[   29.539797] aiy-vision spi0.0: Writing myriad firmware
[   33.905870] aiy-vision spi0.0: Myriad booting
[   34.133019] aiy-vision spi0.0: Myriad ready
```

jcorbett, 2018-06-26:
```
$ dmesg | grep -i aiy*
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:CC:E7:2F vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=96417e6e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Memory: 370980K/393216K available (5950K kernel code, 491K rwdata, 1948K rodata, 396K init, 725K bss, 14044K reserved, 8192K cma-reserved)
[    0.211190] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.350872] hw perfevents: enabled with armv6_1176 PMU driver, 3 counters available
[    0.633723] Waiting for root device PARTUUID=96417e6e-02...
[    3.473459] systemd[1]: Configuration file /opt/aiy/io-mcu-firmware/aiy_io_permission.service is marked executable. Please remove executable permission bits. Proceeding anyway.
[    4.785671] pwm_soft: loading out-of-tree module taints kernel.
[   20.479438] aiy-vision spi0.0: Initializing
[   20.514260] aiy-vision spi0.0: Failed to bind reset GPIO
[   22.830056] aiy-vision spi0.0: Initializing
[   22.839496] aiy-vision spi0.0: Failed to bind reset GPIO
[   22.987784] aiy-io-i2c 1-0051: Setting board type vision
[   22.988784] aiy-io-i2c 1-0051: Driver loaded
[   22.990314] aiy-vision spi0.0: Initializing
[   22.995914] aiy-vision spi0.0: Failed to bind reset GPIO
[   26.224365] pwm-aiy-io pwm-aiy-io: Driver loaded
[   26.225906] aiy-vision spi0.0: Initializing
[   26.227433] aiy-vision spi0.0: Failed to bind reset GPIO
[   27.082936] gpio-aiy-io gpio-aiy-io: Driver loaded
[   27.084484] aiy-vision spi0.0: Initializing
[   27.105298] aiy-vision spi0.0: Resetting myriad on probe
[   27.105315] aiy-vision spi0.0: Resetting myriad
[   27.155541] aiy-adc aiy-adc: Vision bonnet ADC configuration.
[   27.242682] aiy-adc aiy-adc: Driver loaded
[   29.539797] aiy-vision spi0.0: Writing myriad firmware
[   33.905870] aiy-vision spi0.0: Myriad booting
[   34.133019] aiy-vision spi0.0: Myriad ready
```

jcorbett, 2018-06-26:
After looking closer at _drivers/_spicomm it looks like there is a 12M limit? Is this the max size of the model/binaryproto file? I will try downsizing the model and will try again.

dmitriykovalev, 2018-06-26:
@jcorbett If you are using the latest SD card image (the same code is in this repo) then buffer should be reallocated automatically according to the requested size. `Myriad ready` message means everything is fine (please ignore all `Failed to bind reset GPIO`).

jcorbett, 2018-06-26:
@dmitriykovalev the SD card came programmed with it. Is there a way to tell what version is? What is the easiest way to update it to the latest?

dmitriykovalev, 2018-06-27:
@jcorbett Oh, programmed SD card is not the latest, please check here for the latest one: https://github.com/google/aiyprojects-raspbian/releases

jcorbett, 2018-06-28:
@dmitriykovalev updating to the latest image worked - thank you

closed 2018-06-28 by jcorbett

dmitriykovalev, 2018-06-28:
@jcorbett Perfect!