google/aiyprojects-raspbian #417: Error testing on Bonnet the compiled binaryprotobuf for transfer learning model

dvillevald, 2018-06-24:
I followed EXTENSION TO POET TUTORIAL referenced at AIY project website, re-trained mobilenet on my images and compiled the retrained_graph.pb with no (visible) issues on Ubuntu 16.04. However after I uploaded retrained_graph.binaryproto and retrained_labels.txt to Pi and ran (as suggested in step 5):

# On Pi
~/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py \
  --model_path ~/retrained_graph.binaryproto \
  --label_path ~/retrained_labels.txt \
  --input_height 160 \
  --input_width 160 \
  --input_layer input \
  --output_layer final_result \
  --preview 

I got the following error message:
[image]

I am attaching my retrained_graph.binaryproto and retrained_labels.txt:
[retrained_graph.zip](https://github.com/google/aiyprojects-raspbian/files/2131116/retrained_graph.zip)

As suggested at https://github.com/google/aiyprojects-raspbian/issues/335 I updated SD card but it did not help to resolve the issue. 
Thank you in advance for your help.
Dmitri

dmitriykovalev, 2018-06-25:
@dvillevald can you please also attach `dmesg` output after everything fails? There is a chance that Myriad crashes on your model for some reason.

weiranzhao, 2018-06-25:
@dvillevald Are you using the latest bonnet_model_compiler?

weiranzhao, 2018-06-25:
@dvillevald 
I tried to run your compiled model on my vision kit. I can run it. It seems like the model is trained to output 2 classes. 

You said when you compile the default mobilenet model, you got similar error as above. It seems to me that your firmware is still outdated. Could you run this command and let us know the output?
dpkg -l aiy*

Also, how did you update the SD card?

dvillevald, 2018-06-26:
@weiranzhao, @dmitriykovalev  Thanks a lot for helping and pointing out on the probable cause (outdated firmware), much appreciated. I updated SD card image by running

sudo apt-get update; sudo apt-get upgrade; sudo reboot

and it solved the problem so I am closing this issue as resolved.

closed 2018-06-26 by dvillevald