google/aiyprojects-raspbian #418: Vision Kit Not Booting with default CPU frequency

SaltMango, 2018-06-25:
Vision Kit Not Booting with default CPU frequency it only boots if change the  CPU frequency bellow 900MHZ. with default CPU frequency it will freeze at boot with kernel panic error.

dmitriykovalev, 2018-06-25:
Looks like this could be a Raspberry Pi issue. You can try to check without VisionBonnet attached.

SaltMango, 2018-06-26:
Tried that but same error and not booting

dmitriykovalev, 2018-06-26:
Alright, so that must be Raspberry Pi issue then, try to check their forums. Another popular problem is #346 but looks like you have a different one.

burtbick, 2018-07-06:
I just got my Vision kit yesterday and have been fighting to get it to work.  At first I thought it was camera related as it would kernel oops shortly after booting and lock up if the camera was connected.  But I found the other thread talking about the joy app causing the crash so I stopped the joy app as quickly as possible and things kept working.
But obviously that is not a real solution.
Everything is connected correctly, and the camera would work using the raspistill app just fine.
I tried the other suggestion of setting over_voltage in the config.txt file but that didn't resolve the issue.

I found this thread and did some testing.  The comment in the config.txt file says that the default CPU speed is 700 MHz so I tried setting it to 600 and finally I got the buzzer to sound on startup for the very first time and no kernel oops!  Progress.

Next I set the speed explicitly to 700 MHz (supposedly the default value).  And guess what?  It still works and the joy app appears to be recognizing smiles and frowns as it should.

So I then increased the speed to 800 Mhz and then to 900 Mhz and each time things continued to work.

So next I tried 1000 Mhz (1 Ghz the max for the pi w spec).  And at 1 GHz the kernel oops is back.

I then set the speed to 950 Mhz and so far it appears to be working.  So there are a couple of things:

1. The CPU speed is NOT defaulting to 700 Mhz as the comment in config.txt says, but likely defaulting to 1 GHz. That should be OK but...
2. Obviously either this pi w board or the bonnet does NOT like 1 GHz.

So for others who have been chasing this, or similar lockup/booting  issues, try uncommenting the #arm_freq=800 line in /boot/config.txt or go for the gusto and set it to 950 Mhz like this:

arm_freq=950

Hopefully this will help until someone figures out exactly where the problem is.  Given someone else said that they got a different kit and everything worked out of the box this means either there is a problem with the pi w or the video bonnet board that Google should dig into.

I have another pi w that I may try subbing in to see if I can isolate this to the pi w or the bonnet when I get time.  But for now I'm going to let it cook for a while at 950 MHz to see if it continues to work at that speed.