google/aiyprojects-raspbian #420: Error using Vision kit on a clean raspbian setup

maxritter, 2018-06-26:
Hello,

I am trying to run one of the samples of the AIY vision kit on a clean setup of raspbian.

I was able to install the python library and install all other dependencies.

However, when I try to run an image classification demo, I get the following error:

FileNotFoundError: [Errno 2] No such file or directoy: '/dev/vision_spicomm'

So I expect I am missing some service / driver installation, but where do I get it from?

Thanks,
Max

dmitriykovalev, 2018-06-26:
@maxritter We are going to release new image with clear instructions how to setup on clean Raspbian. Regarding your error: you need to install appropriate kernel drivers for the VisionKit. You can get the list of installed AIY debian packages by running `dpkg -l | grep -i aiy`.

nicknauert, 2018-07-23:
So I am running into the same issue. In the meantime until the new instruction set comes down, can you point us to where we could obtain the /dev/vision_spicomm files? They weren't present on the included sd card after following the instructions [here](https://aiyprojects.withgoogle.com/help/#updating-your-sd-card).

nicknauert, 2018-07-24:
@dmitriykovalev I see. I appreciate the quick response! Maybe I'll go back through the build install process and see if it corrects itself.

manoj7410, 2018-08-10:
@nicknauert HACKING.md file has been updated and you can get this file at https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md
This should be helpful in installing AIY packages on clean Raspbian.

dmitriykovalev, 2018-08-10:
@maxritter Please check new instructions! Closing this one for now.

geeksville, 2018-11-03:
@dmitriykovalev @LuwkasLima I think you should possibly reopen this. It seems to me the underlying problem is that aiy-dkms doesn't have proper triggers set in its deb file to force it to reinstall when the kernel is upgraded.  I suspect @LuwkasLima encountered the same thing I just encountered when I did an "apt-get dist-upgrade".  The raspbian folks updated the kernel, so a new kernel came down (yay) but it didn't rebuild and reinstall the aiy-dkms kernel modules.

@LuwkasLima the workaround I used was "apt-get --reinstall install aiy-dkms" to force it.

dmitriykovalev, 2018-11-04:
@geeksville You are right, there are issues with current debian packages, we are going to fix this in the next SD card release which should happen pretty soon.

geeksville, 2018-11-04:
@dmitriykovalev btw your kit is really cool.  I just spent an afternoon yesterday using it to make this twitterbot: [hummingbot](https://twitter.com/hummingbot1).  Photo [here](https://www.thingiverse.com/thing:3195761/files). It worked great and was super easy, if there is any interest I wouldn't mind doing a small writeup.

Keep rocking it!

dmitriykovalev, 2018-11-12:
@geeksville Thanks! Your project looks very cool! Would be nice if you can describe it at https://www.hackster.io/google/products/aiy-vision

geeksville, 2018-11-12:
@dmitriykovalev sure thing here you go: https://www.hackster.io/projects/1295f2.  I'm an ex-googler and if you ever do betas on new hw and want other demos I'd love to give it a shot. (sorry for spamming this issue btw ;-) ).

RootAculus, 2018-11-15:
> @geeksville You are right, there are issues with current debian packages, we are going to fix this in the next SD card release which should happen pretty soon.

@dmitriykovalev My daughter(12 years) is working hard on her Robot Guide Dog which has Aiy-Vision Bonnet. she is experiencing the exact same problem.  

FileNotFoundError: [Errno 2] No such file or directoy: '/dev/vision_spicomm'

She searched online for solutions but no luck at all. Now she is wondering when the new SD Card release shall be available so she can continue to work on her project :) Thanks

Project Pages:
https://selinoid.wordpress.com/2018/10/11/english-coderdojo-coolest-projects-2018-1/
https://www.raspberrypi.org/blog/coolest-projects-international-2018/

dmitriykovalev, 2018-11-17:
@geeksville Thanks for sharing your project on Hackster! Unfortunately for now all new hw testing is internal. Please take a look at https://aiyprojects.withgoogle.com/edge-tpu/ by the way :)

dmitriykovalev, 2018-11-17:
@RootAculus New SD card release is already available https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116  If you are trying to set everything up on the original Raspbian image, please try to follow steps in [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/v20181116/HACKING.md).  

iC4U looks great!

dmitriykovalev, 2018-11-17:
We have updated our debian packages, please try to follow instructions in [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/v20181116/HACKING.md) and `/dev/vision_spicomm` should be available after installing `aiy-vision-dkms` package.

closed 2018-11-17 by dmitriykovalev

dodysw, 2018-11-18:
I must not have done it in the right order, after following HACKING.md (upgrading from August 2018 SD card image release), I had to reinstall the package before `/dev/vision_spicomm` finally appeared.
```
sudo apt install --reinstall aiy-vision-dkms
sudo reboot
```
Thanks for the instructions!

RootAculus, 2018-11-25:
> @RootAculus New SD card release is already available https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116 If you are trying to set everything up on the original Raspbian image, [...]

@dmitriykovalev Thanks again...

She created a new SD Card with the latest released as you suggested.  This time she received an error message that suggests no camera connection.  She tried the camera(v 2.1) on Rpi 3B+ it worked, then she tried a different cable between bonnet and camera, this time camera is recognized but she received the following error message:

 _"No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board_

We checked the connections including the one on the camera, everything seems OK. 

Any recommendations?

Thanks!

dmitriykovalev, 2018-11-30:
@RootAculus Are you trying to run VisionBonnet connected to Pi 3B+ ?  There is an issue here, it doesn't work on 3B+ but it works on 3B if you can find an additional flex cable. Can you try to check that everything works on Pi Zero?

ScenicSwisscoast, 2018-12-02:
same error here. brand new pi vision kit out of the box, and I downloaded the version 181116 and flashed it on a sd-card from here https://github.com/google/aiyprojects-raspbian/releases

I don't manage to get any examples in the folder ~/AIY-projects-python/src/examples/vision up and running

seth-johnson-sp, 2018-12-08:
> same error here: _FileNotFoundError: [Errno 2] No such file or directoy: '/dev/vision_spicomm_'.
> 
> this is with a brand new pi vision kit out of the box, and I downloaded the version 181116 and flashed it on a sd-card from here https://github.com/google/aiyprojects-raspbian/releases
> 
> I don't manage to get any examples in the folder ~/AIY-projects-python/src/examples/vision up and running
> 
> for example: when i run
> 
> > sudo python3 face_camera_trigger.py
> 
> i get this
> _FileNotFoundError: [Errno 2] No such file or directoy: '/dev/vision_spicomm_'.

Same result here. I found that the problem occurs if on first startup I go through the configuration UI and accept the offer to update the software. If I boot from a fresh copy of this image (181116) and choose not to update the Raspbian software, then the AIY vision stuff works. It's the update that triggers the FileNotFoundError: [Errno 2] No such file or directoy: '/dev/vision_spicomm_'. error.

weathermanj, 2018-12-15:
I am seeing the same as the above. Previously I was running on an older image without issue, I upgraded it to the latest today and now I suffer the same as the others. 

> FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'

I have used a new image, and also followed HACKING.md. neither work

sdcharle, 2019-07-28:
Having similar troubles here in July 2019, bummer! No joy at all.

Joseph-L-Hayes, 2019-08-09:
Excitedly got my pi zero w up and running with AIY, all working great except when trying to use any of the demos or camera - I get the same error as everyone else here 

FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'

Any update on this?