google/aiyprojects-raspbian #421: Error compiling binaryproto of SqueezeNet - Facenet

drohack, 2018-06-26:
I'm very new to all of this tensorflow, training, and freezing. I'm attempting to compile the trained squeezenet graph provided in t[his issue](https://github.com/davidsandberg/facenet/issues/613) to run on the Google AIY Vision Kit.

Using the [compiler](https://dl.google.com/dl/aiyprojects/vision/bonnet_model_compiler_latest.tgz) I've tried running this command:
'''./bonnet_model_compiler.par --frozen_graph_path 20180204-160909/20180204-16090.pb --output_graph_path vggface2_sqeezenet.binaryproto --input_tensor_name input --output_tensor_names embeddings --input_tensor_size 160 --debug
''''
But I'm getting this error:

2018-06-24 19:57

> :47.484325: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 1207 operators, 1562 arrays (0 quantized)
> 2018-06-24 19:57:47.526755: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 989 operators, 1360 arrays (0 quantized)
> 2018-06-24 19:57:47.564350: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 989 operators, 1360 arrays (0 quantized)
> 2018-06-24 19:57:47.586820: F external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:46] Check failed: unsupported_ops.empty() These unsupported ops were not removed by graph transformations: Prod, Reciprocal, SquaredDifference, VariableV2

I've tried looking up the `unsupported_ops.empty()` online, but could not parse what was needed to be done to get past this error (or if anything could be done). I'm not even sure if I'm running the compiler on the correct file, or if I need to freeze the graph (or is the 20180204-16090.pb file the already frozen graph), or if the file is in the right format (though it seems to be as it's built on SqueezeNet with a tensor size of 160)...

Thanks for any help

giacomobartoli, 2018-07-10:
@drohack by looking at your error, I would say that the TF graph is using some operators that are not supported by the Vision Kit. Try to inspect the graph through Tensorboard.

weiranzhao, 2018-07-17:
@drohack Aside from the unsupported operators you might be using. SqueezeNet version of FaceNet might be too big to deploy on VisionKit. Even though it is using 160x160 input, it might be using too many channels in intermediate tensors. If it is possible to train a MobileNet based FaceNet, 
input size: 160x160, depth multiplier = 0.5 or
input size: 192x192, depth multiplier = 1.0

It should be straight forward to deploy on VisionKit.

closed 2018-11-17 by dmitriykovalev