google/aiyprojects-raspbian #422: Setup error 403

Mietzsche81, 2018-06-27:
New to the module, so I'm a little lost trying to getting the API up and running.

[I followed the instructions here](https://www.raspberrypi.org/forums/viewtopic.php?t=204865), and am getting the following exception when running `src/main.py`:
```
Traceback (most recent call last):
  File "src/main.py", line 103, in <module>    
    main()
  File "src/main.py", line 97, in main
    with Assistant(credentials) as assistant:
  File "/home/pi/AIY-projects-python/src/aiy/assistant/library.py", line 33, in __init__
    self._model_id = device_helpers.register_model_id(credentials)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/device_helpers.py", line 93, in register_model_id
    r.raise_for_status()
  File "/home/pi/AIY-projects-python/env/lib/python3.5/site-packages/requests/models.py", line 928, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error
```

which links me to a page with the error message:

```
{
  "error": {
    "code": 403,
    "message": "The request is missing a valid API key.",
    "status": "PERMISSION_DENIED"
  }
}
```

I followed the link to allow permissions and pasted the code into the terminal, but that didn't seem to work.

In case it's relevant, during install I received the following warning:
```
Collecting google-assistant-library==0.1.0
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: SysCallError(104, 'ECONNRESET')",),)': /simple/google-assistant-library/
```

and the following error:
`google-api-core 0.1.4 has requirement requests<3.0.0dev,>=2.18.0, but you'll have requests 2.14.2 which is incompatible.`

manoj7410, 2018-06-29:
@Mietzsche81 

Looks like invalid credentials. Can you please follow the steps given at : https://aiyprojects.withgoogle.com/voice/#google-assistant--get-credentials to create credentials for your account (Please make sure that you enable Google Assistant API for your project)? 

Please allow assistant to access your account by enabling all the options here : https://myaccount.google.com/activitycontrols

Additionally, please make sure that the date and time of your PI is matching with current date and time.

BTW, which model of PI are you using ?

dmitriykovalev, 2018-07-28:
@Mietzsche81 let us know if you are still having this problem. Looks like a configuration issue.

closed 2018-07-28 by dmitriykovalev

vinturella, 2019-04-13:
I am also getting the HTTPError. How would I fix the configuration issue?