google/aiyprojects-raspbian #423: Voice demo crashing when trying to speak

mc-coi, 2018-06-28:
I have the Voice 2.0 and everything is installed, speakers are working when I do the sound check (it plays and records and plays my voice). However, when I run the assistant_grpc demo, it listens and determines what I said, then it crashes. I have tried old and new versions of the the code.

I have included a screen shot of the error. Please help, I'm SOOOO close
[image]

manoj7410, 2018-06-29:
@mc-coi Have you tried with the latest software image which is available at : https://aiyprojects.withgoogle.com/help/#updating-your-sd-card

mc-coi, 2018-06-29:
I have. I updated the image and pulled the newest AIY demos. Still having the same issues.

manoj7410, 2018-07-02:
@mc-coi Did you try with more sentences (Like, "Hello", "What is your name")? 

Does it crash immediately after printing the sentence that you say or it waits for few seconds before crashing ?

And can you make sure that your internet connection is not slow ?

dmitriykovalev, 2018-07-28:
@mc-coi SD card image has everything you need to run examples, no need to pull AIY demos separately. Looks like you have a mix of different source code versions. Let us know if problem still exits.

closed 2018-07-28 by dmitriykovalev