google/aiyprojects-raspbian #425: AIY Vision document on clean OS doesn't match source code repo

dlarue, 2018-07-02:
I'm trying to get the vision kit working with my DIYRoboCar software and while trying to use the GPIO PWM in the servo example I'm getting errors regarding PinFactoryFallback and not even finding pins.py to define PIN_A or PIN_B.  

This git repo is missing the install_deps.sh and all I see is requirements.txt so I installed those. I looked at install_services.sh and that's just a call to setup the joy service so of little use/need.

When will the installation document match the code/repo?

`(env) pi@aiy-robocarZW:~/projects/AIY-projects-python $ python src/examples/vision/gpiozero/servo_example.py 
/home/pi/env/lib/python3.5/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/env/lib/python3.5/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
/home/pi/env/lib/python3.5/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from pigpio: failed to connect to localhost:8888
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "src/examples/vision/gpiozero/servo_example.py", line 10, in <module>
    from aiy.pins import PIN_A
ImportError: No module named 'aiy'

`

dlarue, 2018-07-02:
I found a Raspberry Pi forum which talks about installing this on a new OS and it gets an old version of install-deps.sh so I tried it and saw it created it's own virtual env and continued building that. So I sourced it and the servo example ran without errors. I will dissect that and figure out what's the magic sauce to merge DIYRobocar/Donkey with the AIY Vision kit.

manoj7410, 2018-08-10:
@dlarue You can follow updated HACKING.md file from https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md to install AIY packages on a clean OS.

dmitriykovalev, 2018-08-10:
@dlarue  Please checkout new `HACKING.md` as @manoj7410 mentioned. Report new issues for the updated instructions — closing this one now.

closed 2018-08-10 by dmitriykovalev